openthread/openthread #4515: Interest in a new platform example using Linux driver & SPI controlled radio?

PlaidShirtPat, 2020-01-30:
I have previously built a OpenThread platform integration using a SPI controlled radio on a raspberry pi previously for a proof-of-concept, but am not able to contribute that code. I am considering writing a similar implementation from scratch as a personal project, but would like to gauge interest in the concept first.

Here is a high-level outline of my plan:
1. Develop Linux Kernel module to drive a 802.15.4 radio module over SPI (I have a [mrf24j40ma](https://www.microchip.com/wwwproducts/en/MRF24J40MA) for this).
  - This driver/module will expose a API to control the low level 15.4 HW radio functions to user-space applications.
  - Currently, I am considering the IOCTL interface to allow building the module out-of-tree and remove the need for kernel source.
2. Use the current POSIX platform example as a starting point and integrate with the radio driver.
3. Run the new ot-ncp application under wpantund to expose thread networking as a socket. 

Advantages:
- Provides a 'soft' NCP that lets you create a full OpenThread Border Router by only adding a radio module (NCP runs as a process instead on a co-processor).
- Decent cross-compatibility across Linux versions & distributions (the ioctl & SPI interfaces are stable).
- The kernel module can be easily modified to support multiple radio modules with SPI interfaces.

If there is interest in adding something like this, I'll start on it. Knowing that other people could use this functionality would help motivate me :)

abtink, 2020-01-30:
Thanks @PlaidShirtPat for raising the issue to contribute/improve OT functionality :)

I think you are already familiar with [RCP model](https://openthread.io/platforms#radio-co-processor-rcp). A recently merged contribution (PR https://github.com/openthread/openthread/pull/4431) added support for SPI interface between RCP and host (in addition to UART which was originally supported). I think you may be able to re-use/start with this model for your project?

> Develop Linux Kernel module to drive a 802.15.4 radio module over SPI (I have a mrf24j40ma for this).

What are your thoughts on the command format between host and radio module (over SPI) for such a Linux Kernel module? RCP model uses a subset of spinel commands/protocol  to expose low-level radio functions.

PlaidShirtPat, 2020-01-30:
>  you may be able to re-use/start with this [RCP] model for your project?

I'm unsure if the RCP model will work. If I understand correctly, it would require the ability to flash custom open thread firmware on the radio module. I am proposing connecting a "dumb" radio module where low-level 802.15.4 radio configuration & frame transmission is connected over SPI. These modules don't have a programmable micro controller, but a few have driver support in the Linux kernel, but it is through the wpan stack that doesn't expose all of the low-level functionality the open thread stack needs (as of the last time I checked).

> What are your thoughts on the command format between host and radio module (over SPI) for such a Linux Kernel module?

The kernel driver I'm proposing would drive the "dumb" radio module over SPI or another hw bus (different radio models have non-compatible SPI protocols). The API the driver exposes would then be used by the openthread platform layer.

For example, a raspberry pi hooked with a mrf24j radio module wired to the SPI & gpio pins. The 'co-processor' is a user-space process running on the raspberry pi. On startup, the platform code accesses the API provided by the kernel module via IOCTL. The kernel module would handle the low-level radio SPI interactions, and the OT platform code could call driver methods. The driver API would provide the functionality specified in the [OT radio platform API](https://openthread.io/reference/group/radio-operation) like setting panId, ext address, sending & receiving frames.

The reason for the kernel module is because trying to drive a radio module over SPI in a user-space context creates too much delay due to context switching.

abtink, 2020-01-30:
> The kernel driver I'm proposing would drive the "dumb" radio module over SPI or another hw bus (different radio models have non-compatible SPI protocols). The API the driver exposes would then be used by the openthread platform layer.

Having a kernel driver to map the low-level radio function into common API which can then be used as  an OT platform code  sounds reasonable and useful to me. 

I guess the low-level SPI format (between kernel driver and radio) would be vendor/device specific?

PlaidShirtPat, 2020-01-30:
> I guess the low-level SPI format (between kernel driver and radio) would be vendor/device specific?

Exactly.

bukepo, 2020-01-31:
Since Thread is a low rate network protocol, maybe we could start from making a prototype in user space.
Another interesting approach would be running OpenThread based on [linux-wpan](http://wpan.cakelab.org/).

PlaidShirtPat, 2020-01-31:
@bukepo 
> Another interesting approach would be running OpenThread based on linux-wpan.

I've tried linux-wpan in the past and controlling the low level functions through the socket interface wasn't very fast and there were some features I could not use. It has been a while though so that approach may work now. It would be cool if we could re-use the linux network stack 802.15.4/6LOWPAN capabilities.

> Since Thread is a low rate network protocol, maybe we could start from making a prototype in user space.

Trying to control the low level radio functions in user space tends to be pretty slow. The driver code needs to make multiple SPI read & write transactions very quickly for some operations. For example, you may need to:
- send one SPI cmd to query for a radio controller register value
- read the value the radio controller sends back to get the current radio state
- send another command to change to a TX state
- send the 802.15.4 frame to transmit
- toggle a GPIO to trigger sending the frame

There are usually fewer steps, but these are all tasks that the driver code needs to handle in very short time frames.

When you insert a kernel/user-space context switch between each of those steps, the OT network stack running in user space has trouble keeping up with activity on the network (especially as more devices are added). This isn't bad if you have a smart NCP/RCP which only receives the high level spinel commands over SPI and handles the low level radio control.

jwhui, 2020-02-05:
@PlaidShirtPat, thanks for raising this proposal.

In the case of a radio transceiver with fixed, vendor-specific serial interface, I can see the potential advantages of implementing the radio driver in the kernel.

If you proceed down this path, you may want to consider starting with the "posix-host", implemented in [src/posix](https://github.com/openthread/openthread/tree/master/src/posix). There, you would effectively replace the `otPlatRadio*` implementation in [src/posix/platform/radio_spinel.cpp](https://github.com/openthread/openthread/blob/master/src/posix/platform/radio_spinel.cpp). The posix example in [examples/platforms/posix](https://github.com/openthread/openthread/tree/master/examples/platforms/posix) is only used for simulation and continuous integration so far. In contrast, "posix-host" is what we use to run OpenThread on posix-based hosts in real product.

PlaidShirtPat, 2020-02-06:
@jwhui posix-host sounds like a great starting point. Thanks.

jwhui, 2020-07-15:
@PlaidShirtPat , any updates on this efforts?

jwhui, 2020-07-20:
Closing stale issue for now. Please feel free to re-open if/when there is a desire to move this forward.

closed 2020-07-20 by jwhui