openthread/openthread #4523: Silicon Labs Support: Build OpenThread for Silicon Labs as a library and/or import into Simplicity Studio

theHurc, 2020-02-01:
Silicon Labs has apparently dropped support for their proprietary Thread stack in favor of OpenThread. However, the Silicon Labs Eco System revolves around Windows development with their flavor of Eclipse called Simplicity Studio. OpenThread development is geared towards Linux development and building/debugging without an IDE. I don't see any documentation or migration guides for how to either use OpenThread in Simplicity Studio or how to migrate existing applications to OpenThread.

It would be great is Silicon Labs would provide instructions on how to build OpenThread for Silicon Labs as a library and/or import into Simplicity Studio

machatta, 2020-02-05:
Hi ,

Integration into Simplicity Studio is currently being planned.

In the mean-time, please use instructions provided below to build open thread.

You may find the Readme for mg12 at link:
https://github.com/openthread/openthread/tree/master/examples/platforms/efr32mg12

and for mg21 at link:
https://github.com/openthread/openthread/tree/master/examples/platforms/efr32mg21

mg21 is slightly different as a change has to be made to core_cm33.h in our SDK as is mentioned on that page.
 
Best Regards
Azam

closed 2020-02-24 by jwhui

marciogranzotto, 2023-01-18:
updated link:
https://github.com/openthread/ot-efr32