openthread/openthread #4524: Important APIs to commission to Thread Network

bhumikaajain, 2020-02-03:
I am using Nordic SDK for thread and Zigbee V4.0.0 for my application.
In most examples autocommissioning is set to true.
But when I set it to false to commission to my exiting thread network,
Before starting the commissioner at Border Router I get error in JoinerCallback 23 i.e. OT_ERROR_NOT_FOUND 
When I start the start the commissioner at border router, I get joinercallback error 8 i.e. OT_ERROR_SECURITY.
What exactly comprises of OT_ERROR_SECURITY and how can I commission the device and remove this error?

jwhui, 2020-02-05:
@bhumikaajain, given that the "auto-commissioning" feature mentioned here is specific to the Nordic SDK, this may be better asked at https://devzone.nordicsemi.com/, which is the official support portal for nRF5 SDK for Thread.

To give some insight on the errors, `OT_ERROR_NOT_FOUND` typically indicates that the joiner was not able to find a network that is looking to commission the joiner. Note that how the PAN ID is configured before the joining process is started can affect this. In particular, if the PAN ID is set to `0xffff`, the device will look for any network. However, if the PAN ID is not set to `0xffff`, it will only look for networks that match the same PAN ID.

The `OT_ERROR_SECURITY` error simply indicates that the device was not able to establish a DTLS session. This could be because of a mismatch in the PSKd. This could also be due to message delivery failures.

As a general rule, you may want to ensure that non-volatile settings storage is cleared when starting the joining process.

Hope that helps.

wgtdkp, 2020-02-28:
@bhumikaajain `OT_ERROR_NOT_FOUND` **is the expected error** when there is no active commissioner or the joiner has not been added by the commissioner.

bhumikaajain, 2020-02-28:
Hi,

The issue was in Nordic's SDK. It has been resolved.
Thanks for your help.

closed 2020-02-28 by bhumikaajain