openthread/openthread #4530: [autothreadharness] v50 select case doesn't work to choose specific test cases

doublemis1, 2020-02-05:
**Describe the bug**
During execution test ex. Router 5.1.7 autothreadharness crash because of `action_chains.perform()` in harness_case.py file.
It happens every time if any previous test case is not available (not display) in test case tree

**To Reproduce**
Information to reproduce the behavior, including:
1. Openthread master
2. GRL 2.0.0 (v50)
3. Google chrome 80
4. start.bat -t -r 3 --blacklist rfbox_cases -p "Router_5_1_7*.py"

**Console/log output**
```
ERROR:root:case[Router_5_1_7] error
--->    INFO : Traceback (most recent call last):
--->    INFO :   File "C:\Python27\lib\unittest\case.py", line 329, in run
--->    INFO :     testMethod()
--->    INFO :   File "autothreadharness\harness_case.py", line 1106, in test
--->    INFO :     self._select_case(self.role, self.case)
--->    INFO :   File "autothreadharness\harness_case.py", line 839, in _select_case
--->    INFO :     action_chains.perform()
--->    INFO :   File "C:\Python27\lib\site-packages\selenium\webdriver\common\action_chains.py", line 80, in perform
--->    INFO :     self.w3c_actions.perform()
--->    INFO :   File "C:\Python27\lib\site-packages\selenium\webdriver\common\actions\action_builder.py", line 76, in perform
--->    INFO :     self.driver.execute(Command.W3C_ACTIONS, enc)
--->    INFO :   File "C:\Python27\lib\site-packages\selenium\webdriver\remote\webdriver.py", line 321, in execute
--->    INFO :     self.error_handler.check_response(response)
--->    INFO :   File "C:\Python27\lib\site-packages\selenium\webdriver\remote\errorhandler.py", line 242, in check_response
--->    INFO :     raise exception_class(message, screen, stacktrace)
--->    INFO : JavascriptException: Message: javascript error: Failed to execute 'elementsFromPoint' on 'Document': The provided double value is non-finite.
--->    INFO :   (Session info: chrome=80.0.3987.87)
```

**Additional context**
I am not sure if it is problem with new certification tool version or just my environment.

**Fix**
In method `_select_case` add click checkbox with id `display-all-devices`, before getting elements from test case tree.
```
self._browser.find_element_by_id('display-all-devices').click()
checkbox = None
wait_until(lambda: self._browser.find_elements_by_css_selector('.tree-node .tree-title') and True)
```

FiveDimensions, 2020-02-18:
Hi @doublemis1, can you share the webdriver version using cmd "chromedriver version" and check if it is compatible with chrome 80 referring to https://chromedriver.chromium.org/?

doublemis1, 2020-02-18:
My chromedriver version is:
`Starting ChromeDriver 80.0.3987.16 (320f6526c1632ad4f205ebce69b99a062ed78647-refs/branch-heads/3987@{#185})`.
And referring to the website that you provide me my version of chromedriver and chrome is compatible.

FiveDimensions, 2020-02-20:
Ok, just make sure if it is compatible between chrome and chromedriver.  It can not be reproduced using the same autotools, chrome, chromedriver and TH on our testbed. "It happens every time if any previous test case is not available (not display) in test case tree", "previous" means the tested case before current testing case? Can you share the full log file? Thanks

doublemis1, 2020-02-27:
[harness-20200227122356.log](https://github.com/openthread/openthread/files/4261331/harness-20200227122356.log)
[auto-20200227122355.log](https://github.com/openthread/openthread/files/4261332/auto-20200227122355.log)

Logs from Cert tool and autothreadharness.
Also test are executed on widows 10.

FiveDimensions, 2020-03-08:
Hi @doublemis1, 
Thank you for providing the log files. This issue happens on the testbeds with mixed vendor devices using Chrome 80 and Chromedriver 80. The ActionChains.perform() will throw an exception when ActionChains move to the element without text.  Here is a fix PR: 
https://github.com/openthread/openthread/pull/4648
Please have a try on your testbed. Thanks.

jwhui, 2020-03-16:
@doublemis1, can you take a look at #4648 and see if that resolves your issue?

doublemis1, 2020-04-01:
Sorry that it took so long, @FiveDimensions, PR works.

jwhui, 2020-04-01:
> Sorry that it took so long, @FiveDimensions, PR works.

@doublemis1, thanks for testing and reporting back!

FiveDimensions, 2020-04-01:
@doublemis1, cool! thanks for the verification.

closed 2020-04-08 by jwhui