openthread/openthread #4531: ot::Cli::Interpreter Cause Fault when ICMP Echo Reply Received

braiden, 2020-02-06:
**Describe the bug**

If an ICMP echo reply is received after `otCliConsoleInit` is called, but before there's been any user interaction with the CLI, a fault occurs. In my use case such a situation is common. I'm using openthread in zephyr, I start a cli, but rarely interact with it. Someone could craft an ICMP packet to crash my device, and the device crashes whenever I use zephyr's stack to ping.

**To Reproduce**

`otCliConsoleInit` registers an `Interpreter::HandleIcmpReceive` callback in its constructor. This callback tries to do `mServer->OutputFormat()`. `mServer` is null. It only gets a non-null value when `Interpreter::ProcessLine` is called.

Hope that's clear... happy to test.

**Expected behavior**

Don't log ICMP packet when `mServer` is null?

**Console/log output**

**Additional context**

jwhui, 2020-02-06:
@braiden thanks for raising this issue! We’ll work to propose a fix soon.

jwhui, 2020-02-10:
Resolved by #4534.

closed 2020-02-10 by jwhui