openthread/openthread #4535: [adress-query] Address Error Notification

LuDuda, 2020-02-07:
When I enabled all logs during stress tests on 24 NCP-FTD boards (where devices often create different partitions and merge again due to too heavy traffic), I found a lot of this kind of logs:

`Error while processing address error notification: Drop`

Couple conclusions:

1. This particular error is printed since we don't accept non-confirmable CoAP type [here](https://github.com/openthread/openthread/blob/master/src/core/thread/address_resolver.cpp#L567). This is a bug since according to the specification we should be able to parse Address Notification which is sent as a multicast packet with Non-Confirmable CoAP type.

Proposed solution: Just accept NON and CON.

2. The Address Error of the multicast type is only sent in this [line](https://github.com/openthread/openthread/blob/master/src/core/thread/address_resolver.cpp#L470). This means that in the network there at least two devices which share the same ML-EID - which is not the case - another bug.

I spent some time analyzing the code to understand how it could be possible to have such a problem since it does not seem obvious. The only theory I have is that the memcmp checks ML-EID (from Address Notification) with some garbage address. For example, I can imagine, that there was a previous cache entry created by a proactive inspection of mesh header [here](https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder_ftd.cpp#L609), which at the end [created the entry](https://github.com/openthread/openthread/blob/master/src/core/thread/address_resolver.cpp#L269) without any ML-EID modification.

After the stress test, and some stabilization phase, we often saw that certain devices need a couple of ICMPv6 Ping Req/Rsp iterations before they could communicate with all the other devices in the network. I think that the problem was hidden in the Address Resolver module, which actually might be related to the reported above issue.

jwhui, 2020-02-07:
@LuDuda, thank you for identifying this issue and working to resolve it!

I agree we should accept non-confirmable Address Error Notification messages.

I also agree that we need to update the Address Resolver code to handle entries that were created based on snooping messages. In this case, I think we need to make sure `mLastTransactionTime` is set to `kLastTransactionTimeInvalid` in [AddressResolver::AddCacheEntry()](https://github.com/openthread/openthread/blob/c4940b36054e4b0e88c3bc01eacfa7055fb79236/src/core/thread/address_resolver.cpp#L269).

After inspecting the code, I believe the following is possible:

1. A device sends an Address Query.

2. The device receives at least one message from the target before receiving the Address Notification.

3. The device creates a new address cache entry - here we may have two cache entries for the same target. This newly created cache entry does not set the ML-EID nor does it set `mLastTransactionTime`. 

4. The Address Notification is received and the second conflicting entry causes the device to send a multicast Address Error notification.

I don't know if this is exactly what's causing the behavior you see, but I do think it is something we need to address.

Thoughts?

**/cc/** @librasungirl

abtink, 2020-02-08:
Thanks @LuDuda and @jwhui  for nice debugging of the issue.

Reading the code, I agree. I think  we do need to set `mLastTransactionTime` to `kLastTransactionTimeInvalid` in `AddCacheEntry()`. 

One other thing I noticed is whether it is ok to call `MarkCacheEntryAsUsed()`from `AddCacheEntry` (which will put the entry at top of age list) since the entry is not actually used (it's created from snooping a received message). I would say we probably want to set the new entry's age as the last  cached entry (after all entries in cached state but behind entries in invalid/query state)?

Thoughts?

LuDuda, 2020-02-09:
@jwhui @abtink thank you for your thoughts! Please assign me to this issue, the fix is on the way. It looks like I found a couple more bugs in the address resolver module.

PR #4540 (2020-02-11): [address-resolver] resolve several issues
This PR includes four individual commits associated with the address resolver module. It should resolve #4535. It's the easiest to review them one-by-one.

**[address-resolver] accept multicast Address Error Notification**

This commit enables processing of Address Error Notification sent as non-confirmable CoAP multicast message. The code change relaxes the verification of CoAP type.
_____________________________________________________

**[address-resolver] fix last transaction time of entry created by packet inspection**

For cached entries created  [... cut]
closed 2020-02-11 by jwhui