openthread/openthread #4541: [router-table] assert seen while processing RouteTLV

LuDuda, 2020-02-09:
**Describe the bug**
I got [assert](https://github.com/openthread/openthread/blob/master/src/core/thread/router_table.cpp#L518
) in `router_table.cpp` during a stress test of the large network.

I've never seen this one, the only change I did was to enable:
`OPENTHREAD_CONFIG_MLE_SEND_LINK_REQUEST_ON_ADV_TIMEOUT`.
So either this is related to this, or it's a very unlikely situation.

OpenThread commit id (current master): [aa78b2688e72c1bbe66034059f76c6a4f4096e97](https://github.com/openthread/openthread/commit/aa78b2688e72c1bbe66034059f76c6a4f4096e97)

**Console/log output (automatic GDB capture)**
```
0x0001f61e in ot::RouterTable::ProcessTlv (
    this=this@entry=0x20008d54 <ot::gInstanceRaw+33564>, aTlv=...)
    at /openthread/examples/../src/core/thread/router_table.cpp:518
518	            assert(router != NULL);
(gdb) (gdb) #0  0x0001f61e in ot::RouterTable::ProcessTlv (
    this=this@entry=0x20008d54 <ot::gInstanceRaw+33564>, aTlv=...)
    at /openthread/examples/../src/core/thread/router_table.cpp:518
#1  0x0001812c in ot::Mle::MleRouter::ProcessRouteTlv (
    this=this@entry=0x20007840 <ot::gInstanceRaw+28168>, aRoute=...)
    at /openthread/examples/../src/core/thread/mle_router.cpp:1076
#2  0x0001b696 in ot::Mle::MleRouter::HandleLinkAccept (
    this=0x20007840 <ot::gInstanceRaw+28168>, aMessage=..., aMessageInfo=..., 
    aKeySequence=aKeySequence@entry=0, 
    aNeighbor=aNeighbor@entry=0x20008ee4 <ot::gInstanceRaw+33964>, 
    aRequest=aRequest@entry=false)
    at /openthread/examples/../src/core/thread/mle_router.cpp:961
#3  0x0001b70a in ot::Mle::MleRouter::HandleLinkAccept (this=<optimized out>, 
    aMessage=..., aMessageInfo=..., aKeySequence=aKeySequence@entry=0, 
    aNeighbor=0x20008ee4 <ot::gInstanceRaw+33964>)
    at /openthread/examples/../src/core/thread/mle_router.cpp:767
#4  0x00017b24 in ot::Mle::Mle::HandleUdpReceive (
    this=0x20007840 <ot::gInstanceRaw+28168>, aMessage=..., aMessageInfo=...)
    at /openthread/examples/../src/core/thread/mle.cpp:2740
#5  0x00010756 in ot::Ip6::Udp::HandleMessage (
    this=0x200072ac <ot::gInstanceRaw+26740>, aMessage=..., aMessageInfo=...)
    at /openthread/examples/../src/core/net/udp6.cpp:336
#6  0x0000eda8 in ot::Ip6::Ip6::HandlePayload (this=<optimized out>, 
    aMessage=..., aMessageInfo=..., aIpProto=<optimized out>)
    at /openthread/examples/../src/core/net/ip6.cpp:993
#7  0x0000f370 in ot::Ip6::Ip6::HandleDatagram (
    this=0x20007280 <ot::gInstanceRaw+26696>, aMessage=..., 
    aNetif=0x20007360 <ot::gInstanceRaw+26920>, 
    aLinkMessageInfo=aLinkMessageInfo@entry=0x2001ff68, 
    aFromNcpHost=aFromNcpHost@entry=false)
    at /openthread/examples/../src/core/net/ip6.cpp:1225
#8  0x00012cfa in ot::MeshForwarder::HandleDatagram (
    this=this@entry=0x20007764 <ot::gInstanceRaw+27948>, aMessage=..., 
    aLinkInfo=..., aMacSource=...)
    at /openthread/examples/../src/core/thread/mesh_forwarder.cpp:1325
#9  0x000130fe in ot::MeshForwarder::HandleLowpanHC (
    this=this@entry=0x20007764 <ot::gInstanceRaw+27948>, 
    aFrame=0x2000abaf <nrf_802154_rx_buffers+31> "", 
    aFrame@entry=0x2000aba6 <nrf_802154_rx_buffers+22> "\177\063\360MLMLX\260", aFrameLength=88, aFrameLength@entry=97, aMacSource=..., aMacDest=..., 
    aLinkInfo=...)
    at /openthread/examples/../src/core/thread/mesh_forwarder.cpp:1299
#10 0x00013200 in ot::MeshForwarder::HandleReceivedFrame (
    this=0x20007764 <ot::gInstanceRaw+27948>, aFrame=...)
    at /openthread/examples/../src/core/thread/mesh_forwarder.cpp:1039
#11 0x000089ee in ot::Mac::Mac::HandleReceivedFrame (
    this=0x20007650 <ot::gInstanceRaw+27672>, 
    aFrame=aFrame@entry=0x20009e48 <sReceivedFrames>, aError=<optimized out>)
    at /openthread/examples/../src/core/mac/mac.cpp:1744
#12 0x0000993e in ot::Mac::SubMac::Callbacks::ReceiveDone (
    this=<optimized out>, aFrame=aFrame@entry=0x20009e48 <sReceivedFrames>, 
    aError=<optimized out>)
    at /openthread/examples/../src/core/mac/sub_mac_callbacks.cpp:61
#13 0x00009614 in ot::Mac::SubMac::HandleReceiveDone (this=<optimized out>, 
    aFrame=aFrame@entry=0x20009e48 <sReceivedFrames>, aError=<optimized out>)
    at /openthread/examples/../src/core/mac/sub_mac.cpp:196
#14 0x00022a52 in ot::Radio::Callbacks::HandleReceiveDone (
    this=this@entry=0x20004a47 <ot::gInstanceRaw+16399>, 
    aFrame=aFrame@entry=0x20009e48 <sReceivedFrames>, aError=<optimized out>)
    at /openthread/examples/../src/core/radio/radio_callbacks.cpp:43
#15 0x000107f6 in otPlatRadioReceiveDone (
    aInstance=aInstance@entry=0x20000a38 <ot::gInstanceRaw>, 
    aFrame=aFrame@entry=0x20009e48 <sReceivedFrames>, 
    aError=aError@entry=OT_ERROR_NONE)
    at /openthread/examples/../src/core/radio/radio_platform.cpp:49
#16 0x000259c8 in nrf5RadioProcess (
    aInstance=aInstance@entry=0x20000a38 <ot::gInstanceRaw>)
    at /openthread/examples/../examples/platforms/nrf528xx/src/radio.c:652
#17 0x00025d58 in otSysProcessDrivers (
    aInstance=aInstance@entry=0x20000a38 <ot::gInstanceRaw>)
    at /openthread/examples/../examples/platforms/nrf528xx/src/system.c:161
#18 0x000002f8 in main (argc=0, argv=0x0 <__isr_vector>)
    at /openthread/examples/../examples/apps/ncp/main.c:113

(gdb) routerId = 19 '\023'
allocationChanged = true
(gdb) this = 0x20008d54 <ot::gInstanceRaw+33564>
aTlv = @0x2001fb8c: {
  <ot::Mle::Tlv> = {
    <ot::Tlv> = {
      mType = 9 '\t', 
      mLength = 27 '\033'
    }, <No data fields>}, 
  members of ot::Mle::RouteTlv: 
  mRouterIdSequence = 239 '\357', 
  mRouterIdMask = "\r\001%\252A(\210\200", 
  mRouteData = "\002\361\062\000\002\361\302\361\361\001\361\361\062\361\361\361\361\002\000\000\000S\000\000\000\277\000\000\000\027\365\261~\202\000\000\000\366\365\020\063X\000\000\000\340\000\000\000\376\000\000\000\t\000\000\000\t\037\221au"
}
```

**Additional context**
During stress test devices quite often may miss MLE messages due to congestion in the network.

jwhui, 2020-02-10:
@LuDuda, thanks for raising this issue!

At first glance, this condition failure indicates that there is an inconsistency between [mAllocatedRouterIds](https://github.com/openthread/openthread/blob/1edde90d91fff21343bdb8560eafcba6d5cf1331/src/core/thread/router_table.hpp#L357) and [mRouters](https://github.com/openthread/openthread/blob/1edde90d91fff21343bdb8560eafcba6d5cf1331/src/core/thread/router_table.hpp#L356).

1. In every case we update `mAllocatedRouterIds`, we call `UpdateAllocation()` after.

2. The [one place](https://github.com/openthread/openthread/blob/1edde90d91fff21343bdb8560eafcba6d5cf1331/src/core/thread/mle_router.cpp#L977) I know where we call `SetRloc16()` outside of `router_table.cpp` seems to be OK.

I'll need to dig more to see if I can find anything.

LuDuda, 2020-03-29:
Could this happen because of using `OPENTHREAD_CONFIG_MLE_SEND_LINK_REQUEST_ON_ADV_TIMEOUT` where advertisements are missed but link is kept established?

In general, do you see any side effects of using OPENTHREAD_CONFIG_MLE_SEND_LINK_REQUEST_ON_ADV_TIMEOUT in the environment when Advertisements are often lost?

LuDuda, 2020-06-21:
Hello Jonathan,

We haven't seen this assert anymore for a while. Taking to account constantly changing codebase also in regards to the `RouterTable` module, I think we can close it and re-open if such fail reappears.

jwhui, 2020-06-22:
> We haven't seen this assert anymore for a while. Taking to account constantly changing codebase also in regards to the `RouterTable` module, I think we can close it and re-open if such fail reappears.

Thanks, @LuDuda. Will close this for now. Please reopen if you see this again.

LuDuda, 2020-07-03:
And actually we got this assert second time wtih (`OPENTHREAD_CONFIG_MLE_SEND_LINK_REQUEST_ON_ADV_TIMEOUT` enabled). We can do some more investigation since we have more logs.

It seems something like this happened (`Router 0x6400` asserted).
 1. `Router 0x6400` had a couple of Router neighbors and some children.
 2. One of it's FTD child sent Address Solicit and received Response and promoted to `Router 0x3C00`.
 3. `Router 0x6400` received Advertisement from `Router 0xC800`.

```
Route table updated
[INFO]-MLE-----:     1400 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     1800 -> c800, cost:1 1, lqin:3, lqout:3, link:yes
[INFO]-MLE-----:     3c00 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     4800 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     6400 -> ffff, cost:0 16, lqin:0, lqout:0, link:device
[INFO]-MLE-----:     6800 -> c800, cost:1 1, lqin:3, lqout:3, link:yes
[INFO]-MLE-----:     8400 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     b000 -> c800, cost:1 1, lqin:3, lqout:3, link:yes
[INFO]-MLE-----:     c800 -> 6800, cost:1 1, lqin:3, lqout:3, link:yes
```

 4. `Router 0x6400` received Link Request from child and attempted to send Link Accept (this is delayed message).
 5. `Router 0x6400` was reset (due to test scenario - random resets) and was not able to send Link Accept.
 6. `Router 0x6400` came up again and sent Link Request.
 7. `Router 0x6400` received Link Accept from `Router 0x6800` and changed state from Detached to Router.
 8. `Router 0x6400` received Advertisement from `Router 0xC800` (the rloc16 - 0x3C00 is there).

```
[INFO]-MLE-----:     1400 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     1800 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     3c00 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     4800 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     6400 -> ffff, cost:0 16, lqin:0, lqout:0, link:device
[INFO]-MLE-----:     6800 -> ffff, cost:0 1, lqin:3, lqout:3, link:yes
[INFO]-MLE-----:     8400 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     b000 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     c800 -> ffff, cost:0 16, lqin:0, lqout:3, link:no
```

 9. `Router 0x6400` received Link Accepts from `Router 0xB000`, `Router 0x1800` and `Router 0xC800`.

Last log from Router Table update:
```
[INFO]-MLE-----:     1400 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     1800 -> ffff, cost:0 16, lqin:0, lqout:3, link:no
[INFO]-MLE-----:     3c00 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     4800 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     6400 -> ffff, cost:0 16, lqin:0, lqout:0, link:device
[INFO]-MLE-----:     6800 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     8400 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     b000 -> ffff, cost:0 16, lqin:0, lqout:0, link:no
[INFO]-MLE-----:     c800 -> ffff, cost:0 16, lqin:0, lqout:3, link:no
```

 10. `Router 0x6400` received Link Accept from `Router 0x3C00` (old child).

Parsed data from Route TLV included in Link Accept using GDB:
```
0x1800: cost:0, lqin:3, lqout:0
0x3C00: cost:1, lqin:0, lqout:0
0x4800: cost:0, lqin:0, lqout:0
0x6400: cost:1, lqin:3, lqout:3
0x6800: cost:1, lqin:3, lqout:3
0x8400: cost:0, lqin:0, lqout:0
0xA000: cost:0, lqin:0, lqout:0
0xC800: cost:1, lqin:3, lqout:3
```

10. Router removed Router 0x3C00 (old child) from NVM (logs) and asserted.

```
(gdb) info locals
router = 0x0 <_vectors>
routerId = 5 '\005'
allocationChanged = true
```

Do you see any immediate reason for this?

LuDuda, 2020-07-06:
@jwhui @abtink I went through the code, but as well i don't see anything suspicious. It seems like the Router ID 5 was supposed to be created when we received advertisement (then the `Route table updated log` appeared). This test was realized with relatively high data load in the network. Any suggestion from your side is welcome.

simonlingoogle, 2020-07-07:
@LuDuda @jwhui 

I have proposed a fix in #5200. 

**We always need to update existing `router` references  after calling `ProcessRouteTlv`.** Actually we fixed a similar issue in #4860. 

I was able to reproduce this issue in minutes using an [OTNS stress test script](https://github.com/simonlingoogle/ot-ns/blob/debug/4541/pylibs/examples/move_fault.py) in which nodes are constantly moving to inject network failures. This issue do seems only happen when `OPENTHREAD_CONFIG_MLE_SEND_LINK_REQUEST_ON_ADV_TIMEOUT=1`.

With fix #5200, I am unable to reproduce this issue now.

abtink, 2020-07-07:
@simonlingoogle great find! Thanks.

LuDuda, 2020-07-07:
@simonlingoogle this is so awesome! Really appreciate your help.

Also seems that OTNS is really powerful tool :+1:

PR #5200 (2020-07-07): [mle] fix invalid router after ProcessRouteTlv
This PR fixes #4541 

**We always need to update existing `router` references after calling `ProcessRouteTlv`.**

A similar bug was fixed by #4860
closed 2020-07-07 by jwhui