openthread/openthread #4542: [address-resolver] change the age model

abtink, 2020-02-09:
The issue https://github.com/openthread/openthread/issues/4535 triggered a deeper look into the `AddressResolver` implementation.

First let me say, the existing approach and the fix by @LuDuda for the issue all look good and fine.

I was thinking of a possible future enhancement to simplify the logic and replace the age model. 
Describing it here to get feedback on it: 
- We use a cache entry array as a pool.
- We use single linked list to keep the entries in order we want.
  - For linked list we can use a next index (into array) instead of a next pointer to be more memory efficient.
- We can use 4 different lists: 
  - List of cached entries : We keep the list sorted based on order of use - when an entry is used, we move it to the head of the list.
  - List of query-first-time: First time queries (we wait for the timeout).
  - List of query-retry-mode: Queries that failed at least once and are in retry timeout mode. We keep them in order the user requested them.
  - List of unused entries
- This way there is no need to keep state in entry (the list you are in indicates your state). 
- When trying to allocate a new entry, we first check in the unused list, then query-retry-list and finally the cached list from back of the list (oldest entry).

I can help add it later if the approach sounds good.

In the current design I think we may use entries inefficiently in some cases, for example:
- We query for an address, it grabs the highest age one (last one)
- Suppose it times out (no response), we increase failure count and update retry time (age remains the same).
- Now if we query for another address, the same entry (since it's still highest age) would be used even though we may have unused entries we could use.

Thoughts?

jwhui, 2020-02-10:
@abtink, thanks for proposing this enhancement.

I think single-linked list using index to array is a good optimization - it avoids needing to update the age value for all affected entries when inserting/removing entries.

I do **not** think we should reclaim entries that are in `kStateQuery`. The current implementation does not allocate new entries that are in `kStateQuery`: https://github.com/openthread/openthread/blob/b62d42f0ef8cba720d2d90a98a88a322380b6a03/src/core/thread/address_resolver.cpp#L138

Note that in a long-running stable network, I think we should assume that there are no "invalid" entries, since we do not actually have a mechanism that ages out entries (which is intentional).

Also, we may want to consider how we treat entries that were created due to "snooping" messages - see my comment in https://github.com/openthread/openthread/pull/4540#discussion_r376828382.

LuDuda, 2020-02-10:
@abtink I really like your proposal. It would simplify the current implementation a lot! I just followed the current model while bug-fixing.

Also as the developer, I would love to have CLI/NCP command to print the current full cache table (for example to understand connectivity issues).

Since we are talking about address-resolver improvements, I know that the Address Resolver parameters are configurable, but may I ask what do you think about default `ADDRESS_QUERY_INITIAL_RETRY_DELAY` and `ADDRESS_QUERY_MAX_RETRY_DELAY` values being set to 15 seconds and 8 hours. For me, those values seem much too long. Maybe we should just trigger additional automatic Address Query when we don't get a response in a much shorter time, and then just drop the cache entry instead of blocking transmission in the worst case for 8 hours! This is defined in the spec, however, our implementation may also invalidate certain cache entries without possibility to ensure that we won't transmit additional Address Query within the requested time.

abtink, 2020-02-11:
Thanks guys. 

> Also as the developer, I would love to have CLI/NCP command to print the current full cache table (for example to understand connectivity issues).

@LuDuda , I agree it would be good/helpful to expand the NCP/CLI commands. We have commands in NCP/CLI to get the cache table but only valid entries and just address and rloc. In wpantund we have `Thread:AddressCacheTable` property. I think would be good to be able to get the full table, i.e., all entries in all states including query mode with timeout values, etc.

> may I ask what do you think about default ADDRESS_QUERY_INITIAL_RETRY_DELAY and ADDRESS_QUERY_MAX_RETRY_DELAY values being set to 15 seconds and 8 hours. For me, those values seem much too long. 

Fully agree with. The defaults (specially the 8 hours) are too long. IIRC the config parameters were added since the defaults were not really practical and we needed a way to change them in internal project. We use 120 seconds for `MAX_RETRY` (instead of 8 hours), the `INITIAL_RETRY` of 15 seconds is kept as is though.

jwhui, 2020-02-11:
> Fully agree with. The defaults (specially the 8 hours) are too long. IIRC the config parameters were added since the defaults were not really practical and we needed a way to change them in internal projects. We use 120 seconds for `MAX_RETRY` (instead of 8 hours), the `INITIAL_RETRY` of 15 seconds is kept the same.

As @abtink mentioned, we found the existing Thread Specification values too long to provide a good user experience. For background, the concern was around traffic destined for a device that does not exist on the network. In that case, we want something sufficiently long so that repeated Address Query messages do not cause too much traffic overhead. We found 15 seconds and 120 seconds to be much better for our product use cases.

closed 2020-03-31 by jwhui