openthread/openthread #4546: Reconstructing build for an IDE is really difficult

theHurc, 2020-02-11:
I'm building for the EFR32MG12 and I'm trying to reverse engineer the build so I can integrate into an IDE. Running make with the -V option was really helpful for verbose output but there are a few linking issues I am unable to get around. They all appear to be related to mbedtls so I'll just describe one issue and extrapolate from there.

For example, these 2 files are giving me an issue:

"third_party/silabs/gecko_sdk_suite/v2.7/util/third_party/mbedtls/library/ecp.c"
"third_party/mbedtls/repo/library/ecp.c"

They appear to be pretty much identical files. They are built into these static libraries respectively:

libsilabs-efr32mg12-sdk.a
libmbedcrypto.a

However, then magically they are both included in this linker command. (In bold below)

/bin/bash ../../../../../libtool  --tag=CC --preserve-dup-deps    --mode=link arm-none-eabi-gcc  -fdata-sections -ffunction-sections -Os -g -I/mnt/c/Working/openthread/examples/../examples/platforms/efr32mg12//brd4161a -D__START=main -DEFR32MG12P432F1024GL125 -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-efr32-config.h"' -DOPENTHREAD_CORE_CONFIG_PLATFORM_CHECK_FILE='"openthread-core-efr32-config-check.h"' -I/mnt/c/Working/openthread/examples/../examples/platforms/efr32mg12/ -DOPENTHREAD_CONFIG_COAP_API_ENABLE=1 -DOPENTHREAD_CONFIG_IP6_SLAAC_ENABLE=1 -DOPENTHREAD_SETTINGS_RAM=0 -mcpu=cortex-m4 -mfloat-abi=soft -mthumb -Wall -Wextra -Wshadow -Wundef -Werror -Wno-error=undef -std=c99 -pedantic-errors      -T /mnt/c/Working/openthread/examples/../third_party/silabs/gecko_sdk_suite/v2.7/platform/Device/SiliconLabs/EFR32MG12P/Source/GCC/efr32mg12p.ld               -Wl,-Map=sleepy-demo-mtd.map -fdata-sections -ffunction-sections -Os -g -I/mnt/c/Working/openthread/examples/../examples/platforms/efr32mg12//brd4161a -D__START=main -DEFR32MG12P432F1024GL125 -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-efr32-config.h"' -DOPENTHREAD_CORE_CONFIG_PLATFORM_CHECK_FILE='"openthread-core-efr32-config-check.h"' -I/mnt/c/Working/openthread/examples/../examples/platforms/efr32mg12/  -DOPENTHREAD_CONFIG_COAP_API_ENABLE=1 -DOPENTHREAD_CONFIG_IP6_SLAAC_ENABLE=1 -DOPENTHREAD_SETTINGS_RAM=0 -mcpu=cortex-m4 -mfloat-abi=soft -mthumb -nostartfiles -specs=nano.specs -specs=nosys.specs -Wl,--gc-sections   -o sleepy-demo-mtd sleepy_demo_mtd-main.o ../../../../../src/cli/libopenthread-cli-mtd.a ../../../../../src/core/libopenthread-mtd.a      ../../../../../examples/platforms/efr32mg12/libopenthread-efr32mg12.a ../../../../../third_party/silabs/libsilabs-efr32mg12-sdk.a /mnt/c/Working/openthread/examples/../third_party/silabs/gecko_sdk_suite/v2.7/platform/radio/rail_lib/autogen/librail_release/librail_efr32xg12_gcc_release.a               ../../../../../third_party/mbedtls/libmbedcrypto.a  ../../../../../src/core/libopenthread-mtd.a      ../../../../../examples/platforms/efr32mg12/libopenthread-efr32mg12.a ../../../../../third_party/silabs/**libsilabs-efr32mg12-sdk.a** /mnt/c/Working/openthread/examples/../third_party/silabs/gecko_sdk_suite/v2.7/platform/radio/rail_lib/autogen/librail_release/librail_efr32xg12_gcc_release.a               ../../../../../third_party/mbedtls/**libmbedcrypto.a**

Of course my IDE is complaining about all sorts of redefinitions. How could these libraries possibly be linked against one another? Is there something I'm missing here?

theHurc, 2020-02-12:
I was able to figure this out by just removing `third_party/mbedtls/repo/library/ecp.c` from the build but it still doesn't explain why the build was actually linking before

sjlongland, 2020-02-26:
I found running `make` with `V=1` is helpful to gleam out how things are getting called… did that a _lot_ in the early days of getting my build system working for some of the in-house stuff I'm working on (my "IDE" is `fvwm`+`gvim`+`make`+`gcc`+`konsole`… very basic).

Recently, a lot of `./configure` switches got moved to `Makefile` definitions; there's a lot you can learn by perusing `examples/common-switches.mk` and observing what it sets in `CPPFLAGS`.  Once you have worked out what `CPPFLAGS`, `CFLAGS`/`CXXFLAGS` and `LDFLAGS` you need for your target, it should be possible to get your development environment to call `./configure` with the appropriate arguments and environment, then `make` to build the libraries you need.

theHurc, 2020-02-27:
That's exactly what I did. Build with V=1. Thanks!

jwhui, 2020-03-16:
@theHurc, the `ecp.c` file provided by Silicon Labs is intended to replace the one provided by mbedtls. I believe the build with OpenThread's autotools-based system works because the linker inspects the first library when resolving symbols and does not look further after finding a match. Given that `libsilabs-efr32mg12-sdk.a` is listed before `libmbedcrypto.a`, the build completes as expected.

Is there anything you need from this issue?

jwhui, 2020-03-25:
Closing stale issue.

closed 2020-03-25 by jwhui