openthread/openthread #4550: question: non-linux based border router

cfaessler, 2020-02-11:
I am trying to find out whether it is feasible to develop a thread compatible border router which is not based on a linux system. Reason is to save hw costs. We are evaluation the thread technology for a series of products in the smart home area. We don't consider a raspberry pi solution to be a stable product.
Currently it seems that there are hardly an thread compatible border routers available on the market. I wonder what the reason for this is. 
- Does anyone know of any available border router devices on the market?
- Would it be possible to implement a border router without linux? Basically with a TI-1352 chipset and host controller with eth interface.

jwhui, 2020-02-11:
Yes, it is possible to implement a Thread Border Router in a non-Linux environment. The [Nest Connect](https://store.google.com/us/product/nest_connect) is one example of a shipping product that employs OpenThread to implement a FreeRTOS-based Thread Border Router. That said, the Nest Connect does not currently support Thread Commissioning, which makes it difficult to use outside the Nest ecosystem.

You may want to look at the [openthread/ot-rtos](https://github.com/openthread/ot-rtos) project, that provides an integration of OpenThread, FreeRTOS, and LwIP. While the project does not currently implement a Thread/Wi-Fi border router, it should provide a good starting point.

Hope that helps.

jwhui, 2020-03-16:
Closing stale issue.

closed 2020-03-16 by jwhui