openthread/openthread #4560: give up alignment of `\` in Makefiles

bukepo, 2020-02-14:
**Is your feature request related to a problem? Please describe.**
`\` at end of Makefiles are now aligned. However, every time a longer line is added, lines around have to be modified. I don't see big benefits of this alignment style, shall we drop it?

**Describe the solution you'd like**
Just use a single space between `\` and line content.

**Describe alternatives you've considered**
None.

**Additional context**
We usually use $(NULL) to at end of a definition. I think this is for avoiding touch the last item when appending a new item.

bukepo, 2020-02-14:
@jwhui @abtink Thoughts?

bukepo, 2020-02-20:
Ping~

abtink, 2020-02-20:
I don't have any strong preference here but would favor/suggest keeping it as is. I would say this is same as adding a new longer variable name in the code which with our clang-format rule could potentially cause the other definitions/lines to be re-aligned.

bukepo, 2020-02-20:
Thank you for sharing your inputs. Affecting other lines for alignment is not git blame friendly and also increases unnecessary reviewing efforts. As more and more code are added in, I really hope we could do something to improve reviewers experience a little bit.

jwhui, 2020-02-20:
I do not have a strong opinion on the `\` in Makefiles. However, one could also make a similar argument that the same applies to:

- End-of-line comments.
- Multi-line macros.
- Variable declarations.
- Assignment operators.
- param and retval documentation.
- Method declarations.
- ...

Where do we draw the line?

bukepo, 2020-02-22:
Do these styles really improves code readability? Especially

* Multi-line macros.
* Variable declarations.
* Assignment operators.
* Method declarations.

I am fine with requiring alignment for documentation, because we usually read them directly instead of the generation of doxygen. Thoughts?

abtink, 2020-02-24:
I agree and understand that style is subjective but personally I like the code style we currently use in OpenThread. I find the alignments of variable names, assignments, method declarations, multi-line macros,  etc. to make the code look nicer and help with its readability. 

I agree that enforcing such a  code style can cause diffs in PRs/commits to look larger than what the change is and may make it harder for reviewers looking at the diff. But still I would favor enforcing a consistent code style:
- I think the source code as a whole is read more often than any individual PR/commit, so keeping the overall style clean is worth making it slightly harder to look at the diff.
- With `git` we can always use `--ignore-space-change` to see the code change only.
- GitHub also provides same [feature to ignore white-spaces](https://github.blog/2018-05-01-ignore-white-space-in-code-review/).

bukepo, 2020-02-28:
agree.

closed 2020-02-28 by bukepo