openthread/openthread #4581: CLI CoAP URI Query Strings

jordanmelnychukhendrickson, 2020-02-19:
**Is your feature request related to a problem? Please describe.**
Would like to be able to send URI queries with the CoAP CLI command
i.e. `coap get fdde:ad00:beef:0:fdde:ad00:beef:0 api/endpoint?a=1&b=2&c=3`

**Describe the solution you'd like**
In `Coap::ProcessRequest` parse `argv[2]` to split the raw input using `otCoapMessageAppendUriPathOptions` and then iterate over the query params with `otCoapMessageAppendUriQueryOption`

**Additional context**
Was able to send query params to remote device after making changes to cli_coap.cpp described above.
[cli_coap.zip](https://github.com/openthread/openthread/files/4226181/cli_coap.zip)

jwhui, 2020-02-21:
@jordanmelnychukhendrickson, thanks for this proposal. This seems like a good enhancement. Would you like to submit a PR?

jwhui, 2020-03-25:
Closing stale issue.

closed 2020-03-25 by jwhui