openthread/openthread #4597: [meshcop] Commissioner Credential should be used to generate PSKc

edmont, 2020-02-26:
Thread 1.1 Specification section 8.4.1.2 defines how a passphrase (the Commissioner Credential) is used to derive a key (PSKc) that is used to establish a DTLS session between the Commissioner and the Border Agent.

This Commissioner Credential is defined as (8.2):

> A human-scaled passphrase for use in authenticating that a device may petition to become the Commissioner of the network. This passphrase is encoded in utf-8 format and has a length of six (6) bytes minimum and 255 bytes maximum. This credential can be thought of as the network administrator password for a Thread Network.
> The first device in a network, typically the initial Leader, MUST be out-of-band commissioned to inject the correct user generated Commissioning Credential into the Thread Network, or provide a known default Commissioning Credential to be changed later. This credential is used to derive an enhanced key using key stretching called the PSKc (Pre-Shared Key for the Commissioner) which is used to establish the Commissioner Session.

At the moment, I believe OT (API + CLI + docs) is failing to provide this functionality since only the PSKc can be configured when modifying the datasets.

A [tool](https://openthread.io/guides/border-router/tools#pskc_generator) is provided in the OT-BR project to derive the PSKc from the Commissioner Credential, but it's not even mentioned in the OpenThread documentation. And even the Commissioner Credential [is unknown](https://github.com/openthread/openthread/pull/3769) until a _handmade_ PSKc is configured.

In my opinion OpenThread should have an easy way to setup a Thread network with a given Commissioner Credential that can be used later in a Commissioner UI, and it should probably have a default Commissioner Credential in case the user doesn't provide one. At the moment, being able to handle just the PSKc generates a lot of confusion among developers about the terminology, I think. Users shouldn't need to see or use the PSKc at any moment.

jwhui, 2020-02-26:
@edmont, thank you for raising this issue.

I agree we could improve on the use of Commissioner Credential when setting up the network. Perhaps we could enhance the "dataset" CLI command to support generation of datasets with the a Commissioner Passphrase, rather than a random PSKc.

OpenThread already has an API for generating the PSKc [otCommissionerGeneratePskc()](https://github.com/openthread/openthread/blob/91b0dba2a0e212adc9fa2a02860761ec3bada999/include/openthread/commissioner.h#L419), but that is only included when the commissioner feature is enabled.

Of course, in terms of getting/reading datasets, we have no option but to provide the PSKc itself.

I don't think it's a good idea to have a default Commissioner Credential. It's worth noting that many security practices and regulations no longer allow having default credentials that are not device specific.

Would you like to propose some changes?

jwhui, 2020-04-03:
@edmont, does #4766 address this issue?

edmont, 2020-04-06:
@jwhui I believe that the PSKc should be always handled internally in OT, not providing any interface to it. https://github.com/openthread/openthread/pull/4766 is a step forward towards that, but not sure if it allows to close this issue.

However, I understand that removing PSKc handling would lead to backwards compatibility issues.

jwhui, 2020-04-08:
@edmont, I agree that an end-user should not have to deal with handling the PSKc directly. However, I don't think the OpenThread API is the proper place to restrict access to the PSKc. I believe there are valid use cases that benefit from being able to directly access the PSKc. Instead, end-user applications should be responsible for only allowing Commissioner Credential input.

edmont, 2020-04-08:
Yes, that makes sense. Let's close this issue then.

closed 2020-04-08 by edmont