openthread/openthread #4598: [cli] Not possible to set a Network Name using spaces

edmont, 2020-02-26:
The Network Name is defined in Thread as:

> A human-readable utf-8 string to identify the network over-the-air, similar to the Service Set Identifier (SSID) in Wi-Fi. The maximum length of this field is 16 bytes.

However the space character (UTF-8 `0020`) is not possible to be assigned to a Network Name through the [CLI](https://github.com/openthread/openthread/blob/thread-reference-20191113/src/cli/cli.cpp#L1822) since it's used as argument separator.

A possible solution would be to require quotation when entering the Network Name through the CLI.

I found this when trying to reproduce the proposed derivation of PSKc in the Thread 1.1 Specification (8.4.1.2.2), in which "Test Network" is given as example of Network Name.

jwhui, 2020-02-26:
@edmont, thank you for raising this issue.

I agree it would be nice to support space characters in the Network Name. Would you like to submit a PR with the proposed change?

simonlingoogle, 2020-02-27:
This is an issue I am aware of recently. 
I think there are generally 3 options:

1. Use quotation for CLI arguments that contain whitespaces
    - e.x. `networkname "hello world"`
2. Use windows style `\<space>` to escape whitespaces in commands
    - e.x. `networkname hello\ world`
3. Add new command that receives a hex network name
    - e.x. `networknamex 68656c6c6f20776f726c64`

Option 1 and 2 would general solutions for all commands, meanwhile option 3 only handles commands like `networkname`, `domainname` specifically.

Thoughts? @librasungirl @jwhui

edmont, 2020-02-27:
I would say the first option is cleaner for the user.

But notice that values of Network Name containing quotes should be allowed as well, so the logic would be: a single string argument is the one between the two farthermost quote characters. This disallows commands taking two consecutive UTF-8 string arguments, which I don't think are needed now anyway.

In that sense, option 2 is more versatile.

edmont, 2020-03-04:
Fixed by https://github.com/openthread/openthread/pull/4608.

closed 2020-03-04 by edmont