openthread/openthread #4611: Discussion: Semantic of OT_ERROR_PARSE and OT_ERROR_INVALID_ARGS

wgtdkp, 2020-02-28:
This PR is opened for discussion of the semantic of `otError`, specifically, `OT_ERROR_PARSE` and `OT_ERROR_INVALID_ARGS`.

https://github.com/openthread/openthread/blob/c36fa453f1726f4e5f810264ede54a56b75252aa/include/openthread/error.h#L90-L98

It says that `OT_ERROR_PARSE` stands  for both message and arguments parsing error, while **I think `OT_ERROR_PARSE` should only represents message parsing error**. **Because arguments parsing error is a sub-category of invalid arguments error while message parsing error is another error category with the same level of invalid arguments error.**

Current semantic leading to confusing that We can't tell if there are problems with the arguments or we has received malformed messages. `INVALID_ARGS` errors are often caused by the user who doesn't understand the API or CLI argument requirements well. When there is argument parsing error, returning `OT_ERROR_INVALID_ARGS` clearly tells the caller/user of the API/CLI that they are providing bad arguments and the user got a chance to fix it, while returning `OT_ERROR_PARSE` doesn't indicates whether the user is using the API/CLI wrong, or there is internal message processing error.

To make the conclusion, I think arguments parsing error is invalid arguments error (errors that can be avoided) and must be corrected by the user of the API or CLI, while message parsing error has its own category (errors that can't be avoided) and openthread should continue working when it happens. Thus, **I think we can't mxied them into `OT_ERROR_PARSE`**.

Thoughts? @jwhui @librasungirl @bukepo @abtink  @simonlingoogle @gjc13 @Irving-cl

Irving-cl, 2020-02-28:
It makes sense to me. I think **OT_ERROR_INVALID_ARGS** is better for API/CLI.

jwhui, 2020-02-28:
All sounds good to me. Thanks for proposing this.

bukepo, 2020-02-29:
In the current implementation, not recognized diag commands will output string like

`diag feature 'what-ever' is not supported`

With #4556, this error is represented with error OT_ERROR_NOT_IMPLEMENTED. Are we going to mark all CLI  errors as invalid arguments?

In my option, commands are commands, arguments are arguments. Using different error codes (result in different error message) is more user friendly.

Besides, with #4608, users may input a line that can not even be correctly split into arguments. For example, `invalidescape\here`. Do we also output invalid arguments error even before validating arguments? I think in this case, OT_ERROR_PARSE is better because it's parsing stage.

Thoughts?

wgtdkp, 2020-02-29:
@bukepo I think your ideas make sense. And I did give an example case there:

>2. For openthread CLI, a user may input a command that cannot be recognized by the CLI. it is better to define and return OT_ERROR_INVALID_COMMAND other than OT_ERROR_PARSE. Conceptually, INVALID_COMMAND errors are one kind of error which should and can be corrected by the user (just like INVALID_ARGS).

We should define more specific error codes to improve readability for users (`OT_ERROR_PARSE` will lead to confusing). How do you think about this?

librasungirl, 2020-02-29:
Maybe `OT_ERROR_NOT_FOUND` be used for commands not supported?

wgtdkp, 2020-02-29:
@librasungirl Thanks for proposing, but I think it will lead to confusing too.
https://github.com/openthread/openthread/blob/8ab37df1a7c7099dddbe9005f9f7e27fab189ff1/include/openthread/error.h#L176-L179

For example, suppose there is a CLI command `bbr primary` that gives information of current Primary BBR but user inputs `bbbr primary`. Returning `OT_ERROR_NOT_FOUND` will confuse the user that they don't known whether there is something wrong with his input or there is no active primary BBR.

librasungirl, 2020-02-29:
When no primary bbr, the output would be :)
> bbr
Bbr Primary: None
Done

However agree that it may cause confusion for the CLI user to interpret it - whether an unsupported command or no content for the command in some scenarios.

simonlingoogle, 2020-02-29:
Agree that `OT_ERROR_INVALID_COMMAND` is more clear than `OT_ERROR_PARSE` for wrong commands. 

However, I think returning `OT_ERROR_PARSE` for invalid commands also make sense. OT's CLI implementation can be logically split into two parts:
- **Parsing**: Parse a line into a command and arguments. Each command has its `Process` function. 
- **Executing**: Call the `Process` function with arguments. Each `Process` function has its own arguments validation implementation. 

From this perspective, when a user inputs a wrong command (e.x. `udp drink ff02::1 hello`), the `Parsing` procedure will fail and return `OT_ERROR_PARSE`, which correctly reflects what happened. It should not return `OT_ERROR_INVALID_ARGS` because the argument validation procedure hasn't began yet. 

Thoughts @wgtdkp

wgtdkp, 2020-03-03:
@simonlingoogle 
> From this perspective, when a user inputs a wrong command (e.x. udp drink ff02::1 hello), the Parsing procedure will fail and return OT_ERROR_PARSE, which correctly reflects what happened.

I think returning `TO_ERROR_INVALID_COMMAND` works for this case. Agree?

simonlingoogle, 2020-03-03:
> @simonlingoogle
> 
> > From this perspective, when a user inputs a wrong command (e.x. udp drink ff02::1 hello), the Parsing procedure will fail and return OT_ERROR_PARSE, which correctly reflects what happened.
> 
> I think returning `TO_ERROR_INVALID_COMMAND` works for this case. Agree?

Yes, I am **for** adding new error code `TO_ERROR_INVALID_COMMAND` to clearly tell users of unknown commands.

closed 2020-03-19 by jwhui