openthread/openthread #4623: some cmake builds fail

abtink, 2020-03-02:
Current master 15f6808206b233586 trying to build ncp with cmake I get the following error.
```
[337/402] Building CXX object src/ncp/CMakeFiles/openthread-rcp.dir/ncp_base_ftd.cpp.o
FAILED: src/ncp/CMakeFiles/openthread-rcp.dir/ncp_base_ftd.cpp.o 
/usr/bin/c++  -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=1 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=1 -DOPENTHREAD_CONFIG_NCP_UART_ENABLE=1 -DOPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_1 -DOPENTHREAD_EXAMPLES_SIMULATION=1 -DOPENTHREAD_POSIX=1 -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"../tests/toranj/openthread-core-toranj-config.h\" -DOPENTHREAD_RADIO=1 -DOPENTHREAD_SPINEL_CONFIG_OPENTHREAD_MESSAGE_ENABLE=0 -DPACKAGE_NAME=\"OPENTHREAD\" -DPACKAGE_VERSION=\"20191113-00356-g15f680820\" -D_BSD_SOURCE=1 -D_DEFAULT_SOURCE=1 -Ietc/cmake -Iinclude -Iexamples/platforms/simulation -Isrc -Isrc/core -Wall -Wextra -Wshadow -Wno-c++14-compat -fno-exceptions -std=gnu++98 -MD -MT src/ncp/CMakeFiles/openthread-rcp.dir/ncp_base_ftd.cpp.o -MF src/ncp/CMakeFiles/openthread-rcp.dir/ncp_base_ftd.cpp.o.d -o src/ncp/CMakeFiles/openthread-rcp.dir/ncp_base_ftd.cpp.o -c src/ncp/ncp_base_ftd.cpp
In file included from src/core/coap/coap_secure.hpp:35,
                 from src/core/meshcop/commissioner.hpp:42,
                 from src/ncp/ncp_base_ftd.cpp:54:
src/core/meshcop/dtls.hpp:39:10: fatal error: mbedtls/certs.h: No such file or directory
   39 | #include <mbedtls/certs.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```
To reporuce use `./tests/toranj/build.sh cmake`.

abtink, 2020-03-02:
The failing build seems to be RCP (openthread-rcp).
I think the PR https://github.com/openthread/openthread/pull/4604 exposed this issue.

jwhui, 2020-03-02:
Yes, this is a result of #4604 and moving the include paths into `third_party/mbedtls/CMakeLists.txt` https://github.com/openthread/openthread/blob/15f6808206b2335862cf4aaec7ee2babe627a6f2/third_party/mbedtls/CMakeLists.txt#L44

I believe that's why we had the mbedtls include paths specified in the root `CMakeLists.txt` file before.

@bukepo, thoughts on how to address?

abtink, 2020-03-02:
I have a fix in mind. 

We can exclude the `ncp_base_ftd.hpp` and `ncp_base_mtd.cpp` files in the RCP build. I think this will work.

jwhui, 2020-03-02:
> We can exclude the `ncp_base_ftd.hpp` and `ncp_base_mtd.cpp` files in the RCP build. I think this will work.

Yes, I think this works. Thanks for the fix.

jwhui, 2020-03-03:
Resolved by #4624

closed 2020-03-03 by jwhui