openthread/openthread #4630: [platform] Separate transport-related functions from platform.

piotrkoziar, 2020-03-03:
**Is your feature request related to a problem? Please describe.**
Transport is an integral part of the platform.
All functions that initialize and process UART, SPI transport are currently called inside the platform-dependant implementation of the following functions:

`otSysInit`
`otSysDeinit`
`otSysProcessDrivers`

 from the `openthread-system.h` header (located in `examples/platforms`).
 
 NCP and CLI are using the implementation of transport functions from the platform library. 

**Describe the solution you'd like**
The transport layer is separated from the platform and is linked to program in independent library:
1. UART, SPI initialization and process functions are called from the new system functions, for example:
`otSysTransportInit`
`otSysTransportDeinit`
`otSysTransportProcessDrivers`

In addition to `otSysInit` and `otSysProcessDrivers`, user calls `otSysTransportInit` and `otSysTransportProcessDrivers` to use UART/SPI transport in the application.

2. All transport-related functions are in separate precompiled library.

3. Transport headers (`uart.h`, `spi-slave.h`) are moved from `include/openthread/platform` to `include/openthread/transport`.

**Describe alternatives you've considered**
Moving transport to a new library does not have to be done for every platform.

jwhui, 2020-03-09:
@piotrkoziar, thank you for raising this issue.

For background, the `otSys` prefix denotes functions that are neither used nor provided by the OpenThread library. The `otPlat` prefix denotes functions that are used to interface between the OpenThread library and platform-specific code.

Can you explain further the motivation and rationale for separating out UART/SPI functions into a separate namespace and library? Also the motivation and rationale for moving transport-specific headers to `include/openthread/transport`?

Thanks.

piotrkoziar, 2020-03-12:
@jwhui Thank you for your answer.

The idea is to eliminate the need for rebuilding of OpenThread platform libraries when modification of transport-related functions is required. In particular, the rebuild is unwanted when the OpenThread libraries has been already certified.

In other words, separating out UART/SPI functions into a separate library gives the freedom of defining the transport functions without changing the Thread certified stack.

jwhui, 2020-03-13:
@piotrkoziar, that makes sense to me. Thanks for the clarification!

jwhui, 2020-03-25:
@piotrkoziar, was this issue addressed by #4673 or are you going to propose more changes?

piotrkoziar, 2020-03-25:
@jwhui #4673 implemented #4630 to some degree - transport-related functions are delivered in separated library.  
However #4630 proposes further changes: 
* moving transport initialization and processing from system.c to application code (e.g. otSysTransportInit). As a result, the transport will be completely separated from the platform (and certified code).
* moving headers with transport function declarations to separate folder.

Points above require modification of each platform code. I can try to prepare PR with proposed changes.

jwhui, 2020-03-25:
@piotrkoziar, thanks for the update.

Regarding moving headers to a separate folder, you had proposed `include/openthread/transport`, what about `include/openthread/platform/transport` give that these are platform APIs?

piotrkoziar, 2020-03-25:
@jwhui Agree, that sounds reasonable.

jwhui, 2020-06-05:
@piotrkoziar , any updates on this effort?

piotrkoziar, 2020-06-08:
@jwhui We are no longer interested in implementing this. The platform - transport relation as it is now is sufficient. Maybe in the future, there will be a good case for implementing this solution.

closed 2020-06-08 by jwhui