openthread/openthread #4641: Long list of ChildInfo entries stored on flash

kbogucki, 2020-03-05:
We found an issue where node entered a bootloop state.

**Our setup:**
- Devices based on NRF52840
- Network containing ~120 devices
- All nodes connected to cloud via Border Router with NCP

Issue appeared on one device after ~2 weeks of operation.

The device was resetting just after bootup, while calling `otInstanceInitSingle()`. During investigation we found that there are 270 `ChildInfo` entries in openthread settings stored on flash. In `MleRouter::RestoreChildren()` there is an iteration over all `ChildInfo` items. This iterator uses `mIndex` which is `uint8_t` and in our case its value wraps causing infinite loop.
As a solution I change `mIndex` to `uint16_t` and now it works. See my pull request https://github.com/openthread/openthread/pull/4640.

Anyway could you please let me know if storing such a long list of `ChildInfo` is correct?

abtink, 2020-03-05:
@kbogucki , thanks for raising the issue and the fix. The fix you from https://github.com/openthread/openthread/pull/4640 looks fine to me.

Can you please share the value of `OPENTHREAD_CONFIG_MLE_MAX_CHILDREN` config in you build (i.e. how many children are allowed?).

We have the following recovery code in `MleRouter::RestoreChildren(void)` where during init, we check if the child entries in non-volatile have any duplicate or there are more entries than max allowed, if so we call `RefreshStoredChildren()`.
https://github.com/openthread/openthread/blob/225f9b58eb4573e349d785eb022bba192ebc7dd6/src/core/thread/mle_router.cpp#L3610-L3619
And  `RefreshStoredChildren()` should clean all the child info saved in non-volatile  and re-write the entries again. 
https://github.com/openthread/openthread/blob/225f9b58eb4573e349d785eb022bba192ebc7dd6/src/core/thread/mle_router.cpp#L3655-L3668

If the `OPENTHREAD_CONFIG_MLE_MAX_CHILDREN` config is larger than 270, then I would say it may be possible to have 270 entries in non-volatile (though would still be very unlikely since in your set-up there were 120 nodes total).

Nopik, 2020-03-05:
`OPENTHREAD_CONFIG_MAX_CHILDREN` (this is older OpenThread, not master), is set to `100`

jwhui, 2020-03-05:
> `OPENTHREAD_CONFIG_MAX_CHILDREN` (this is older OpenThread, not master), is set to `100`

Can you provide the git commit id?

Nopik, 2020-03-05:
Yeah, I believe it is the one from Nordic's SDK 2.0: https://github.com/openthread/openthread/commit/41d8e4a73b4de4bea96804ea9685d25d477805c6

simonlingoogle, 2020-03-06:
Hi, I once came across an issue that `RemoveStoredChild` failed to remove child info. But I can't reproduce now. Maybe it is fixed by #4234 . Will investigate more.

simonlingoogle, 2020-03-06:
I can reproduce the issue only after reverting PR #4234, as shown below:
```
info	2020-03-06 13:21:50.818 - Node<3> - [INFO]-CORE----: Non-volatile: Added ChildInfo {rloc:0xf4ad, extaddr:4ac150fd0d1191c4, timeout:240, mode:0x0f, version:2}
info	2020-03-06 13:21:50.818 - Node<3> - [INFO]-CORE----: Non-volatile: conflict ChildInfo {rloc:0xf416, extaddr:1a1f7939d9f9a5f6, timeout:240, mode:0x0f, version:2}
info	2020-03-06 13:21:50.818 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf417, extaddr:1ec815914d255830, timeout:240, mode:0x0d, version:2}
info	2020-03-06 13:21:50.818 - Node<3> - [INFO]-CORE----: Non-volatile: conflict ChildInfo {rloc:0xf422, extaddr:1a1f7939d9f9a5f6, timeout:240, mode:0x0f, version:2}
info	2020-03-06 13:21:50.820 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf424, extaddr:2e6981aec62efaa7, timeout:240, mode:0x0f, version:2}
info	2020-03-06 13:21:50.820 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf4a3, extaddr:7660bad0ce11230f, timeout:240, mode:0x05, version:2}
info	2020-03-06 13:21:50.820 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf4a4, extaddr:463b0e11953d5389, timeout:240, mode:0x0d, version:2}
info	2020-03-06 13:21:50.820 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf4a5, extaddr:2e0dc289ca26d214, timeout:240, mode:0x05, version:2}
info	2020-03-06 13:21:50.821 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf4a8, extaddr:327de4b8fecb53d7, timeout:240, mode:0x05, version:2}
info	2020-03-06 13:21:50.821 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf4a9, extaddr:aa3886270e2adb62, timeout:240, mode:0x0f, version:2}
info	2020-03-06 13:21:50.821 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf4aa, extaddr:22feed3c765fb730, timeout:240, mode:0x0d, version:2}
info	2020-03-06 13:21:50.821 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf4ab, extaddr:aa1adee17b29ec63, timeout:240, mode:0x0d, version:2}
info	2020-03-06 13:21:50.821 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf4ac, extaddr:9aa7d37f180faabc, timeout:240, mode:0x0d, version:2}
info	2020-03-06 13:21:50.821 - Node<3> - [INFO]-CORE----: Non-volatile: list ChildInfo {rloc:0xf4ad, extaddr:4ac150fd0d1191c4, timeout:240, mode:0x0f, version:2}
```
**There are duplicate `ChildInfo` entries for child `extaddr=1a1f7939d9f9a5f6` with different RLOC16s**, one of which should have been removed. 
As time goes, these invalid entries might accumulate, leading to more than 255 ChildInfo entries. 

@Nopik Can you check if #4234 works for you?

kbogucki, 2020-03-06:
IMO the problem with current implementation is that `RefreshStoredChildren()` is called only during initialization (not periodically in runtime) so duplicated `ChildInfo` entries can grow up until next initialization (after device reboot), which is dangerous especially when device is rebooted rarely.
@simonlingoogle It’s hard to verify in a short time how your fix affects our case, since it reproduces very rarely. However if your change prevents storing duplicate `ChildInfo` entries in runtime then I think it's a good step forward.

simonlingoogle, 2020-03-06:
> @simonlingoogle It’s hard to verify in a short time how your fix affects our case, since it reproduces very rarely. However if your change prevents storing duplicate `ChildInfo` entries in runtime then I think it's a good step forward.

Yes. Especially now we have fix #4640 , the invalid ChildInfos can be cleaned automatically when node is restarted. My suggestion is that you can deploy two versions of nodes, one with #4234 and one without, and monitors the number of stored `ChildInfo` entries on these nodes to check if it will increase beyond `OPENTHREAD_CONFIG_MLE_MAX_CHILDREN`.

jwhui, 2020-03-25:
Closing for now. Feel free to reopen if the issue is not resolved.

closed 2020-03-25 by jwhui