openthread/openthread #4646: RLOC 0x0000 causing problems?

mwilniewiec, 2020-03-07:
We have a huge hardware setup, currently 120 NRF52840 end nodes + 1 NCP and I observed some strange behaviors while having 0x0000 RLOC.

1. Our NCP registers as a service in a leader and normally we can see 0D TLV in a network data with a valid NCP RLOC16 in it. We have only 1 service so normally there is only one service TLV in netdata #4472 .
2. NCP became a router and assigned itself a 0x0000 RLOC ( is 0 actually a valid Router ID? )
3. Until the NCP changed its RLOC to other value (became a child of some other router) I could observe strange things in network data. Netdata was containing multiple service (0D) TLVs with strange RLOCs and 0x0000 RLOC didn't appear there as usually.
4. I have a feeling that having a 0x0000 RLOC can cause more issues but currently Im not able to give you more specifics.

librasungirl, 2020-03-08:
@mwilniewiec 

0 is a valid router id. The valid router id range is [0, 62]

[A sample test](https://github.com/librasungirl/openthread/tree/test_service_rloc0000) for server (provide service or prefix) with router id 0 shows that thread network data could have the right server field 0x0000.
[image]

It might be some issue occurring in the scale of your physical setup.  Would you please share the captures around the Leader and the Server?

jwhui, 2020-03-09:
@mwilniewiec, can you also indicate what git commit you are using?

mwilniewiec, 2020-03-11:
@jwhui quite old Nordic's SDK 2.0: 41d8e4a with multiple different patches.
RLOC 0x0000 is a rare case, but I'll try to reproduce that with some special code (like @librasungirl did) and verify If I see any strange behaviour and do deeper investigation.

mwilniewiec, 2020-03-26:
As you showed that issue is not reproducible in OT and we are planning to move to a newer version of OT very soon we will do the tests there. If Im able to find a reproducible path I'll create a separate issue. Closing for now.

closed 2020-03-26 by mwilniewiec