openthread/openthread #4649: [efr32] Entropy questions

dismirlian, 2020-03-08:
Hi all. I have some questions regarding the efr32 platforms' entropy implementation.

1. Why is the ADC used instead of the TRNG in efr32mg12 and efr32mg13? I know that there is an erratum associated with the xg13 chip revisions A~C, but not from D onwards. The mg21 platform abstraction uses the TRNG. I'd like to propose to use the TRNG for all platforms, with the ADC only as fallback. This should free the ADC for the application to use.

2. The entropy collection code for the mg21 is as follows:
https://github.com/openthread/openthread/blob/fdcb8553b12216d53ed3c5249a97ebe7ff2e10b7/examples/platforms/efr32mg21/entropy.c#L40-L63
The `otPlatEntropyGet` function discards 3 out of 4 entropy bytes. Why? Couldn't we call directly `mbedtls_hardware_poll`?

3. In the mg21 code, shouldn't we check `ouputLen`? Specifically, seeing the mbedtls/sl_crypto/trng.c file, the `mbedtls_trng_poll` call could fail, setting `outputLen` to 0.

Thanks!

nigelg-silabs, 2020-03-11:
On point 1 - We kept the ADC implementation already present in OT when we released the MG12 and MG13. This is something we plan to change to use TRNG - as we have on the MG21.

On point 2 - We followed the behaviour that was already present on the MG12 Open Thread implementation. Our accelerator has no length limit so we could change this in the future if it makes sense.

On point 3 - Although the Hardware accelerator could fail it is most unlikely to do so. However we agree that this check should be in place to catch this error and will raise a ticket internally to address this.

dismirlian, 2020-03-12:
Hi @nigelg-silabs, thanks for your answer.

1. Great. I thought that there could be more fundamental reasons why not to use the TRNG on mg12 and mg13.
2. Ok, I think that we could do without the `randomUint32Get` intermediate function.
3. Ok, great.

Maybe we could address 2 and 3 after the efr32 platform unification?

Thanks!

romacdon, 2020-09-08:
@dismirlian - sorry for the delay on this.  I didn't realize this issue existed until just today.  I must have missed the reassignment back in July.  Anyway, I agree with @nigelg-silabs's assessment and we will look to address these issues in an upcoming PR.  We are pursuing the platform unification and also integration with our latest GSDK.

dismirlian, 2020-09-11:
Hi @romacdon no problem, great to hear about the platform unification.

PR #5744 (2020-10-30): [efr32] call mbedtls_hardware_poll directly in otPlatEntropyGet()
Closes #4649
closed 2020-10-30 by jwhui