openthread/openthread #4650: [settings] Latency and flash usage/wear questions

dismirlian, 2020-03-08:
Hi, I have some questions regarding the non-volatile OT settings.

### Question 1

MCU platforms do flash write accesses (page erase or word write) with different strategies:
  - Some need to disable interrupts and run the flash write/erase function from RAM.
  - Some can run the flash write/erase function from flash with interrupts enabled, but stall the flash accesses, so if the IVT and processing code is in flash, the interrupts are also stalled.
  - Some have two or more flash banks, and can erase or write to one bank while executing code from another bank.

In the first two cases, interrupt latency is heavily impacted. For instance, my platform's page erase time is ~40ms.

I have another thread running in parallel, independent of the OT thread, which performs a time-sensitive operation, and can't be blocked for 40ms. However, this thread runs very sporadically, and for 1 or 2 seconds. I can't place that code in RAM, because it is rather large.

My question is: can I block the OT thread for 1 or 2 seconds in case OT needs to do a flash write/erase while the other thread is running? What would be the consequences, besides increased response time/packet loss? Do I need to "tune" some of the OT parameters?

### Question 2

This one is related to flash usage and wear.

How often should I expect OT to commit some setting to flash? Are all the settings mandatory? I'm asking, specifically, about `ParentInfo` and `ChildInfo`, which I suppose can change frequently in some scenarios (also, newbie question: why are these required to be non-volatile?).

```
    enum Key
    {
        kKeyActiveDataset     = 0x0001, ///< Active Operational Dataset
        kKeyPendingDataset    = 0x0002, ///< Pending Operational Dataset
        kKeyNetworkInfo       = 0x0003, ///< Thread network information
        kKeyParentInfo        = 0x0004, ///< Parent information
        kKeyChildInfo         = 0x0005, ///< Child information
        kKeyReserved          = 0x0006, ///< Reserved (previously auto-start)
        kKeySlaacIidSecretKey = 0x0007, ///< Secret key used by SLAAC module for generating semantically opaque IID
    };
```
Also, `NetworkInfo` seems to encompass some "fast" changing information, such as RLOC16, alongside some "almost-fixed" info, such as the extended address and the IID.

Bottomline: what can I do to avoid saving to flash as much as possible?

Thank you!

jwhui, 2020-03-09:
> My question is: can I block the OT thread for 1 or 2 seconds in case OT needs to do a flash write/erase while the other thread is running? What would be the consequences, besides increased response time/packet loss? Do I need to "tune" some of the OT parameters?

Blocking the OT thread itself can cause extra delays in responding to request messages from neighboring devices. This may affect neighbors trying to establish new child-router or router-router relationships. Occasionally affecting these is probably OK, but will mean that it could take longer for devices to establish links during these periods.

If blocking the OT thread also involves block the 802.15.4 radio's ability to receive and acknowledge data frames, this can be more problematic. This would make the device look as if it "disappeared" from the network for 1-2 seconds. Neighboring devices trying to send packets to it would not receive ACK frames and eventually declare the link as "dead", causing devices to reattach or routers to propagate new routing topology information. For this, you could look into modifying:

- https://github.com/openthread/openthread/blob/c54238fdd612cce027bc7b095954f0229b031608/src/core/config/mac.h#L73

- https://github.com/openthread/openthread/blob/c54238fdd612cce027bc7b095954f0229b031608/src/core/config/openthread-core-default-config.h#L343

> Bottomline: what can I do to avoid saving to flash as much as possible?

In a stable topology, devices should only be writing to flash to maintain the MAC and MLE frame counters. OpenThread trades between the number of frame counter values to skip on reboot vs. frequency of writes during normal operation:

- https://github.com/openthread/openthread/blob/c54238fdd612cce027bc7b095954f0229b031608/src/core/config/openthread-core-default-config.h#L229

Things like Child and Parent info could be viewed as optimizations - maintaining Child and Parent info in non-volatile allows devices to reattach to the network with low overhead (fewer message exchange) and helps maintain the existing topology (avoids other issues due to topology changes). That said, I believe maintaining Child and Parent info in non-volatile is currently required to achieve Thread certification.

Hope that helps.

dismirlian, 2020-03-09:
Hi @jwhui thanks for your answer!

> Blocking the OT thread itself can cause extra delays in responding to request messages from neighboring devices. This may affect neighbors trying to establish new child-router or router-router [...]

Ok, that's clear.

> If blocking the OT thread also involves block the 802.15.4 radio's ability to receive and acknowledge data frames, this can be more problematic. This would make the device look as if it [...]
In my case, the radio will be capable of buffering and ACKing the received packets, so I shouldn't expect these issues...

> In a stable topology, devices should only be writing to flash to maintain the MAC and MLE frame counters. OpenThread trades between the number of frame counter values to skip on reboot vs. frequency [...]
Ok, I will investigate that configuration. In a non-stable topology, I imagine that OT will store also the Child/Parent info and the Network Info (because of RLOC), right?

> Things like Child and Parent info could be viewed as optimizations - maintaining Child and Parent info in non-volatile allows devices to reattach to the network with low overhead (fewer message exchange) 

I imagined this would be the case...

> and helps maintain the existing topology (avoids other issues due to topology changes). 
What kind of issues would there be if Child and Parent infos are not saved?

>That said, I believe maintaining Child and Parent info in non-volatile is currently required to achieve Thread certification.

Ok.

> 
> Hope that helps.

Yes, thank you very much!

jwhui, 2020-03-09:
> In a non-stable topology, I imagine that OT will store also the Child/Parent info and the Network Info (because of RLOC), right?

A router will store/delete Child info whenever a child attaches or is removed. A child will store parent info whenever it attaches to a new parent. A device will store network info whenever it changes its role (child, router, leader) or the MAC/MLE frame counters increase past the window.

> What kind of issues would there be if Child and Parent infos are not saved?

Not saving Child and/or Parent info will cause the child to perform a full reattach after the child and/or parent resets. If the child does not reattach to the same parent, the EID-to-RLOC cache will need to be refreshed for packets to be sent to the proper parent.

Hope that helps.

dismirlian, 2020-03-10:
@jwhui, very clear. Thanks for your help! Closing.

closed 2020-03-10 by dismirlian