openthread/openthread #4667: [EFR32] Recipe for target 'do-build-cortex-m4' failed

grja, 2020-03-10:
I have updated my code from commit 1edde90 to latest b5b97ae43, and since then build failed for Silabs efr32mg12 platform, board BRD4166A.

**To Reproduce**
Build steps:
- Download and install the GNU toolchain for ARM Cortex-M
- In a Bash terminal:
```console
$ cd <path-to-openthread>
$ ./script/bootstrap
```
- Download and install the Simplicity Studio.
- Install Flex (Gecko) SDK including RAIL Library from Simplicity Studio.
- Configure the path to Flex SDK source code.
```
$ cd <path-to-openthread>/third_party
$ mkdir silabs
$ ln -s <path-to-Simplicity-Studio>/developer/sdks/gecko_sdk_suite silabs/gecko_sdk_suite
```
- Build OpenThread Firmware on EFR32 platform.
```
$ cd <path-to-openthread>
$ ./bootstrap
$ make -f examples/Makefile-efr32mg12 BOARD=BRD4166A
```
More details about build configuration:
[https://github.com/openthread/openthread/tree/master/examples/platforms/efr32mg12](url)

**Console/log output**
```
  BUILD    efr32mg12
make -j10 -C build/efr32mg12 --no-print-directory all
  CHECK         .local-version
  GEN           .local-version
make --no-print-directory all-recursive
Making all in include
make  all-recursive
Making all in openthread
Making all in platform
make[6]: Nothing to be done for 'all'.
make[6]: Nothing to be done for 'all-am'.
Making all in third_party
Making all in silabs
make[4]: *** No rule to make target 'gecko_sdk_suite/v2.7/hardware/kit/common/bsp/bsp_bcc.c', needed by 'gecko_sdk_suite/v2.7/hardware/kit/common/bsp/libsilabs_efr32mg12_sdk_a-bsp_bcc.o'. Stop.
make[4]: *** Waiting for unfinished jobs....
Makefile:485: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:601: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
Makefile:527: recipe for target 'all' failed
make[1]: *** [all] Error 2
examples/Makefile-efr32mg12:293: recipe for target 'do-build-cortex-m4' failed
make: *** [do-build-cortex-m4] Error 2
```

nigelg-silabs, 2020-03-11:
Our build servers aren't showing an issue with the 4166 build and we have successfully built the 4166 on our machines this morning using the latest sdk release and latest git master.

We are wondering why you have done the following as that directory should exist with code in it:
$ cd <path-to-openthread>/third_party
$ mkdir silabs

grja, 2020-03-11:
You are right. I used wrong path name.
Thank you for quick response.

closed 2020-03-11 by grja