openthread/openthread #4680: MAC Histograms

MarekPorwisz, 2020-03-13:
Hi all, 

We are considering adding some more MAC layer histograms similar to the current RetrySuccessHistogram. Currently it is enabled using OPENTHREAD_CONFIG_MAC_RETRY_SUCCESS_HISTOGRAM_ENABLE define. 

Would you prefer us to create a separate enable define for each histogram type (..RSSI_HISTOGRAM_ENABLE, ..SIZE_HISTOGRAM_ENABLE, etc.) or would it be better if all of them would be under the same define and enabled all together(something like ..DIAGNOSTIC_HISTOGRAMS_ENABLE)?

Thanks,
/\/\arek

abtink, 2020-03-13:
Since these are all at MAC layer and somewhat related (info about rx/tx) and I think unlikely that we would need to enable one without the other ones, I would favor/suggest using same config option.
It'd also help reduce the number of possible config option combinations we may want to build and test.

jwhui, 2020-03-13:
Agree with @abtink. Unless we have a good reason for introducing separate build options, I prefer using the same build config.

tkons, 2020-03-16:
But one histogram is not on MAC layer:
MLE_ACCESSORY_RSSI_HISTOGRAM

tkons, 2020-03-16:
Maybe lets use separate CONFIGS for each statistic (with common HISTOGRAM suffix), and lets activate all them using one CONFIG ? Morty would be pleased, and we would have clean code

closed 2020-06-05 by jwhui