openthread/openthread #4684: [doc] configuration description missing in openthread.io

bukepo, 2020-03-14:
**Is your feature request related to a problem? Please describe.**
I couldn't find OpenThread's configurations description in https://openthread.io/reference. I think it helpful to have a centered place to introduce all OpenThread configuration and usage.

**Describe the solution you'd like**
Add a generated configurations page.

jwhui, 2020-03-14:
Related to #4038

Vyrastas, 2020-03-16:
If we want to do something like this right, it should be hosted on GitHub (does it need to be per platform?) as a Markdown file, and we can import it into openthread.io.  So the source would be editable by anyone whenever the switches change (how often does that happen?) and I just need to run a quick script to update the website.

jwhui, 2020-03-16:
> If we want to do something like this right, it should be hosted on GitHub (does it need to be per platform?) as a Markdown file, and we can import it into openthread.io. So the source would be [...]

We already have doxygen in header files in [src/core/config](https://github.com/openthread/openthread/tree/master/src/core/config). Would it be possible to autogenerate docs using the config header files as source?

Vyrastas, 2020-03-16:
Yeah that would be easier.  I assume we'd use all the header files in there?

jwhui, 2020-03-16:
> Yeah that would be easier. I assume we'd use all the header files in there?

Yes, I think so.

Vyrastas, 2020-03-16:
I'll take a look at this

jwhui, 2024-01-12:
Configuration variables documentation is available here: https://openthread.io/reference

closed 2024-01-12 by jwhui