openthread/openthread #4689: [mac] inefficient use of radio in SED mode on nRF52840

e-rk, 2020-03-16:
**Describe the bug**
While conducting current measurements it was observed that during SED operation the radio on nRF52840 is inefficiently used, because before the data request, the OT stack explicitly requests the radio to enter RX state. This consumes extra power and is unnecessary on nRF52840 platform.

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id: `8a1992e2e42fb707babe9491a6a1456e553490e8`, present on master as well
2. IEEE 802.15.4 hardware platform: nRF52840
3. Build steps: any configuration
4. Network topology: One parent and one SED

**Expected behavior**
The RX state is requested only if the platform requires it.

**Console/log output**
Current data request implementation
[image]

Expected behaviour:
[image]

**Additional context**
After commenting out the `Get<Radio>().Receive` call in `void SubMac::BeginTransmit(void)` the charge required to do the data request was lower by 30%, as seen on the second picture.

abtink, 2020-03-17:
`radio.h` platform defines the following state machine with regards to radio transitions and how OT core would/should use radio platform APIs:
https://github.com/openthread/openthread/blob/bf1cf97b0ffbf77d52bc8c9346e3280f5bbaf845/include/openthread/platform/radio.h#L236-L248
Per this, we would need to go to "receive" first before we can "send".   

The  [`otPlatRadioTransmit()`](https://github.com/openthread/openthread/blob/bf1cf97b0ffbf77d52bc8c9346e3280f5bbaf845/include/openthread/platform/radio.h#L554-L571) also mentions in its documentation that it can be called when radio is  "receive" state and should return `OT_ERROR_INVALID_STATE ` if called from other states.  I think some other radio platforms expect and require this pattern/behavior, so would not recommend changing this (in the core common Mac/SubMAC `BeginTransmit()`).

I am wondering why calling `otPlatRadioReceive()` on nRF52840 would cause this longer wait time before tx starts. Doesn't the `otRadioReceive()` return quickly (or does it take some action)?

e-rk, 2020-03-17:
The difference between those two scenarios is that in the first case, the we are entering the RX, which requests the HFCLK to ramp-up ASAP and we begin receiving. This lasts for the duration of the CSMA/CA backoff. After the backoff we make an attempt to transmit the frame.
In the second case the HFCLK is kept turned off until the CSMA/CA backoff elapses.

I have made a proposal PR https://github.com/openthread/openthread/pull/4697
Let me know if the approach with adding a capability for this case is okay with you.

jwhui, 2020-03-17:
@e-rk, thank you for raising this issue and submitting a proposal!

I have reviewed #4697 and this looks good to me.

abtink, 2020-03-17:
@e-rk , thanks for explaining the difference.
The approach of adding a new capability sounds good to me.

jwhui, 2020-03-17:
Resolved by #4697.

closed 2020-03-17 by jwhui