openthread/openthread #4710: [flash] improve robustness to power failure

jwhui, 2020-03-19:
**Describe the bug**
As reported by @dismirlian in https://github.com/openthread/openthread/pull/4552#issuecomment-592943770

> If I understand the code correctly, I think there may some small problems regarding power-failure robustness:
> 
> 1. We always assume that the record with `kFlagAddBegin` set will also have its `mLength` correctly set. If the firmware fails to write the whole header (for instance, if it resets after writing `mFlags`), then the record is left in an inconsistent state. For example, `mFlags` with `kFlagAddBegin` but `mLength` erased (=65535). This would have negative side effects, because it impacts all the methods that iterate over the records, specifically, the `Init` method only checks for `record.IsAddBeginSet()` and then reads the length:
>    https://github.com/openthread/openthread/blob/381f4deda9a6a3d986261046fc2618743ac2422d/src/core/utils/flash.cpp#L69-L73
>    
>    After `Init` initializes `mSwapUsed` incorrectly, all other methods would iterate over garbage.
>    The same happens in `Swap`.
> 2. We always assume that the space after the last record is "writeable" (that is to say, it is erased). This may not be the case if the `mKey` was written but the `mFlags` wasn't.
> 
> A possible solution to both issues would be to enhance the `Init` method to "sanitize" the settings area:
> 
> * If a record is found whose `mFlags` has `kFlagAddBegin` but not `kFlagComplete`, we can assume an interrupted operation, should not count that record in `mSwapUsed`, and trigger a `Swap`. This should get rid of the invalid record.
> * After the last (valid) record, `Init` should scan the remaining space in the swap block and trigger a `Swap` if any byte is != 0xff. This will ensure that the remaining space is "writable".

jwhui, 2020-03-19:
@dismirlian, would you like to submit a PR to address this issue?

dismirlian, 2020-03-19:
Hi @jwhui, yes, no problem. I'm currently preparing a PR to fix a problem with the new efr32 flash platform driver.

I can submit another one addressing this issue afterwards.

jwhui, 2020-03-19:
> Hi @jwhui, yes, no problem. I'm currently preparing a PR to fix a problem with the new efr32 flash platform driver.
> 
> I can submit another one addressing this issue afterwards.

@dismirlian, great, thanks!

closed 2020-03-26 by jwhui