openthread/openthread #4722: [flash] assertion fail in platform/simulation/flash.c

wgtdkp, 2020-03-22:
**Describe the bug**
The OT commissioner CI triggers a openthread assertion fail in simulation/flash.c:
https://github.com/openthread/openthread/blob/5a82d90e4df891d06dfb649d7674aa70a019bea5/examples/platforms/simulation/flash.c#L80

**To Reproduce**
rerun this OT Commissioner CI job https://travis-ci.org/github/openthread/ot-commissioner/jobs/664722719

**Expected behavior**
No assertion failure.

**Console/log output**
https://travis-ci.org/github/openthread/ot-commissioner/jobs/664722719#L5964

**Additional context**
OT Commissioner pulls latest openthread master commit for testing.

jwhui, 2020-03-23:
@wgtdkp, thanks for raising this issue.

It seems to be related to permissions specific to the particular ot-commissioner test. This is the output when adding `perror` to the code:
```
starting meshcop joiner...
spawn /tmp/test-ot-commissioner/ot-cli-ftd-non-ccm 2
factoryreset
flash open: Permission denied
```
Thoughts?

wgtdkp, 2020-03-24:
@jwhui The cause is: we created the `tmp` when running wpantund as root and later trying to create flash file under `tmp` when starting CLI program without root-privilege. I submitted https://github.com/openthread/ot-commissioner/pull/44 in ot-commissioner to fix the CI failure. But, I think we should allow `otPlatFlashInit` to fail, and upper layer should handle the failure.

jwhui, 2020-03-24:
@wgtdkp, thanks for identifying the root cause.

Can you provide more detail on your proposal? If `otPlatFlashInit()` returns an error, what would the upper layer do?

wgtdkp, 2020-03-24:
@jwhui I think we should propagate the error up to `otInstanceInitSingle` and return `nullptr` for failures. How do you think?

jwhui, 2020-03-24:
> @jwhui I think we should propagate the error up to `Instance::InitSingle` and return `nullptr` for failures. How do you think?

What would the caller do once `nullptr` is returned? In the case of the example apps, they will probably just die anyway.

wgtdkp, 2020-03-24:
>In the case of the example apps, they will probably just die anyway.

Yes, In the case of the example apps, they will die. But for those use OT as a library, they may print a message like "initializing OT failed, exting...".

To my understanding, non-volatile storage is a mandatory for Thread and an implementation should stop working if can't even initialize the storage (as what we do now). The assertion here assume we can always succeed to init flash, but is it naturally impossible for a failure during flash initialization? To my understanding, a flash initializer can fail.

Another problem with the assertion is that it is by default removed in release build, in which case we can't report the error and OT will continue as it is working, which hides the error.

On the other hand, by propagating error up to `Instance::InitSingle`, we have a chance to give the detail at the user-facing point if we can implement [error-with-message](https://github.com/openthread/ot-commissioner/pull/45) or something like `perror` in the future.

LuDuda, 2020-03-24:
Not related fully, but after the new flash has been merged, we found the assert in nRF528xx as well (in a different location). We're going to fix it ASAP.

jwhui, 2020-03-24:
> Not related fully, but after the new flash has been merged, we found the assert in nRF528xx as well (in a different location). We're going to fix it ASAP.

@LuDuda, thanks!

jwhui, 2020-03-24:
> Yes, In the case of the example apps, they will die. But for those use OT as a library, they may print a message like "initializing OT failed, exting...".
>
> To my understanding, non-volatile storage [...]

The assertion identified in this issue is specific to simulation. This is also platform-specific driver code. I'd rather the platform deal with any unexpected errors in platform code directly.

In simulation, I'd rather know exactly when and where something unexpected occurs rather than trying to handle it gracefully. `assert()` does that quite well for me.

wgtdkp, 2020-03-25:
>The assertion identified in this issue is specific to simulation. This is also platform-specific driver code.

Not only the simulation platform flash driver can fail, but also those real-board flash drivers, for example, `kw41z` and `samr21`.
https://github.com/openthread/openthread/blob/35bb0027d0b6d2b1b2e54a289bb3e3f56def57d0/examples/platforms/kw41z/flash.c#L55-L60

`FLASH_Init` returns some status code and we silently drop it.

> I'd rather the platform deal with any unexpected errors in platform code directly.

I think the platform driver has no ability to handle it, but the user have. For applications using openthread, they can use some LEDs (or even LCD screen) to indicating the status (for example red LED1 for initialization error) and they can push some re-start button to reset peripherals. We currently handle assertion failure on real board with dead loop, but wouldn't it be hard for user to find out the dead loop by themselves and finally knows the initialization failed?

> In simulation, I'd rather know exactly when and where something unexpected occurs than trying to handle it gracefully. assert() does that quite well for me.

I am ok to use `assert` in simulation code to die and find the failure position, since simulation is not considered serious.

jwhui, 2020-03-25:
> Not only the simulation platform flash driver can fail, but also those real-board flash drivers, for example, `kw41z` and `samr21`.
> https://github.com/openthread/openthread/blob/35bb0027d0b6d2b1b2e54a289bb3e3f56def57d0/examples/platforms/kw41z/flash.c#L55-L60
> 
> `FLASH_Init` returns some status code and we silently drop it. it seems we should assert here?

FWIW, these are _example_ drivers. In many cases, this is not code that may be considered "production quality".

> I think the platform driver has no ability to handle it, but the user have. For applications using openthread, they can use some LEDs (or even LCD screen) to indicate the status (for example red LED1 for initialization error) and they can push some re-start button to reset peripherals. We currently handle assertion failure on real board with dead loop, but wouldn't it be hard for user to find out the dead loop by themselves and finally knows the initialization failed?

A properly-implemented platform could handle this completely outside of the OpenThread library. If `FLASH_Init` has the possibility to return error, one option is to have the platform call `FLASH_Init` before initializing the OpenThread instance.

> I am ok to use `assert` in simulation code to die and find the failure position, since simulation is not considered serious.

In general, code in `examples/` should be considered as such. If there is platform code that should be elevated to "production quality", we can consider moving it out of `examples/`.

wgtdkp, 2020-03-25:
> In general, code in examples/ should be considered as such. If there is platform code that should be elevated to "production quality", we can consider moving it out of examples/.

okay, then current implementation works for me. Thanks for the clarification!

closed 2020-03-25 by wgtdkp