openthread/openthread #4732: 2 Devices attain leader role in a single thread network without Partition

bhumikaajain, 2020-03-25:
We have a small setup of the Raspberry Pi based Border Router (which has 1 NCP and 1 CLI device attached to its 2 USB ports) , 5 FTDs and 10 SED forming a thread network.

Hardware Platform: **Nordic's nrf52840 chip.**
Openthread Commit ID: https://github.com/openthread/openthread/commit/8a1992e2e42fb707babe9491a6a1456e553490e8.
We have built the libraries with following macros enabled..
` make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COAP=1 COMMISSIONER=1 DISABLE_BUILTIN_MBEDTLS=1 DNS_CLIENT=1 DIAGNOSTIC=1 EXTERNAL_HEAP=1 JOINER=1 LINK_RAW=1 MAC_FILTER=1 MTD_NETDIAG=1 SERVICE=1 UDP_FORWARD=1 ECDSA=1 SNTP_CLIENT=1 COAPS=1 DHCP6_SERVER=1 DHCP6_CLIENT=1 USB=1 CHILD_SUPERVISION=1`

SDK used: nRF5 SDK for Thread and Zigbee v4.0.0

Border router commit ID: 345bf7d

We have used NCP and CLI firmware directly from Nordic's SDK and modified FTDs and SEDs as per our application. We have disabled the CLI functionality in all FTDs and SEDs by default and have enabled child supervision in FTDs and SEDs firmware.

Poll period in SEDs: 30 seconds.
Child timeout : 60 seconds.
Child Supervision interval in FTD: 129
ChildSupervisionCheckTimeout in SED: 190

The problem I am seeing is that all FTDs take up the Leader role.
How I know that all have leader role, well in my application I am calling the function "otThreadGetDeviceRole()".The command returns role and based on that we have state variable which can have any of the 3 values - Child, Router and Leader. When the role is disabled or detached, it means that the device is not connected to the thread network and no data Border router would get from that device; hence no change in state variable of our application. So the state variable of all FTDs is showing LEADER data value when they all are in the same thread network.
At the same time when I run leaderdata command on CLI at Border Router, it says  "Leader router ID - of only one FTD"

Please suggest how to debug this, how can multiple FTDs send their state as Leader at the same time.

jwhui, 2020-03-25:
Have you configured the operational dataset (which includes master key, channel, PAN ID, etc.) to the same value on each device? Can you provide the exact steps you used to configure each device so that someone else can reproduce?

bhumikaajain, 2020-03-25:
As I have mentioned earlier, I have used Nordic's SDK for thread and Zigbee V4.0.0.

In application, In thread_instance_init, I have set autocommissioning to false.
After every reboot, device check its status with otDatasetIsCommissioned, If it is commissioned then application proceeds further to send data to border router else joiner is started with PSK.
At border router I enable the commissioner using the same PSK.
The device gets commissioned on the same network once commissioner is started at border router.

Commissioning process is a one time process, device saves network details.

All devices are on Network Channel: 25 and rest of the network details are same on each device.

Along with data FTDs send their device role (child, router or leader) as well to the border router. Device role is fetched in FTD firmware using otThreadGetDeviceRole. My problem is that on CLI at Border Router when I run leaderdata command, it gives leader router ID of only one FTDs.
But inside FTD firmware,  otThreadGetDeviceRole API is returning OT_DEVICE_ROLE_LEADER in multiple FTDs in the same network. 
So as per data received at border router from FTDs, multiple FTDs are in leader state at the same time.

jwhui, 2020-03-25:
FTDs form their own Thread partition when they are not able to communicate with each other.

Are you able to obtain the Active Operational Dataset from the FTDs and ensure that they match among the different devices? You can use [otDatasetGetActive()](https://openthread.io/reference/group/api-thread-general#otdatasetgetactive) to retrieve the Active Operational Dataset.

If the Active Operational Datasets match, are you able to provide a [packet sniffer](https://openthread.io/guides/pyspinel/sniffer) trace?

bhumikaajain, 2020-03-25:
The FTDs are in the field set-up and I can't modify anything on them at present.

As per your statement, FTDs form their own Thread partition when they are not able to communicate with each other.
Does this means that they will form their network partition even when they are able to communicate to same Border router.

Because when I run `wpanctl get` command on border router, In router table I can find eui64 of all the FTD devices on that network including those which consider themselves as leader. 
And if their active operational Dataset is different can they send their data to same Border router ?

jwhui, 2020-03-25:
FTDs should not form their own partition if they are able to communicate directly with each other.

Can you clarify whether the FTDs are able to communicate with the border router? For example, can you ping the FTDs from the border router? If you are able to ping the FTDs, then they must be in the same Thread partition. Devices in different Thread partitions cannot generally communicate IPv6 messages between themselves.

bhumikaajain, 2020-03-25:
Yes FTDs are able to communicate with the border router and I am able to ping the FTDs from the Border Router.

jwhui, 2020-03-25:
If your FTDs are able to communicate with the border router, they must be in the same Thread partition. Leaders choose their own partition IDs and cannot communicate directly with another Leader. Could there be a bug/issue in how the device role is being reported? A [packet sniffer](https://openthread.io/guides/pyspinel/sniffer) trace could help confirm that the devices are in fact within the same partition or not.

bhumikaajain, 2020-03-26:
I have again checked in my application code and found a condition which was preventing the state update which I have fixed now. FTDs are giving their correct role now.
Thanks for your help!

jwhui, 2020-03-26:
> I have again checked in my application code and found a condition which was preventing the state update which I have fixed now. FTDs are giving their correct role now.
> Thanks for your help!

Glad you were able to find the issue!

closed 2020-03-26 by jwhui

nikhil30081995, 2021-06-23:
> I have again checked in my application code and found a condition which was preventing the state update which I have fixed now. FTDs are giving their correct role now.
> Thanks for your help!

Hello, Can you suggest me which part of the code did you find he issue ? I am facing same issue. The RCP device connected to border router is my leader, but when i add a FTD device to network it mention a state of Leader as well. 

From ot-ctl terminal, i am able to ping the mesh link of new ftd device.