openthread/openthread #4741: Mesh local address communication fails eventually

mwilniewiec, 2020-03-26:
We observe a strange behavior where our nodes stop sending packets to NCPs Mesh local address. This is a rare case but as we have 140 hardware nodes setup it happens lets say once a day on a single device. Power reboot fixes the issue and normal communication comes back. Before the reboot we can see problem persisting for days.

Deeper investigation on a failed node showed that:
1. it is alive, and CLI works fine
2. sending packets to the same destination using RLOC instead of Mesh Local works fine
3. sending packets to the same destination using ALOC works fine
4. sniffer showed that the node is not even trying to send anything when Mesh Local is used as a destination. No packets translated to RLOC nor /a/aq to get the RLOC from ML
5. all other (139) nodes send packets to the same Mesh Local address just fine

I did tests with ping command sent from CLI and a sniffer sitting close to a problematic device.
> ping fd3c:9b17:1634:0:0:ff:fe00:4c00   (pcap frame 104739)
> 16 bytes from fd3c:9b17:1634:0:0:ff:fe00:4c00: icmp_seq=23 hlim=64 time=31ms
> 
> ping fd3c:9b17:1634:0:fd33:96eb:becc:8097
> 
> ping fd3c:9b17:1634:0:fd33:96eb:becc:8097
> 
> ping fd3c:9b17:1634:0:fd33:96eb:becc:8097
> 
> ping fd3c:9b17:1634:0:0:ff:fe00:4c00  (pcap frame 108632)
> 16 bytes from fd3c:9b17:1634:0:0:ff:fe00:4c00: icmp_seq=30 hlim=64 time=38ms

Attaching sniff screen and all commands executed in CLI. I've got full 12MB, 30 min sniff from that setup and there was a big mess (lots of reparenting and connection problems) during that time. This could be also interesting for you. Can share on a separate channel if needed.

[image]

[no_ml_commnads_used.txt](https://github.com/openthread/openthread/files/4389877/no_ml_commnads_used.txt)

jwhui, 2020-03-26:
@mwilniewiec, thank you for raising this issue.

I wonder if this is related to the EID-to-RLOC cache. Are you able to provide the output of the [eidcache](https://github.com/openthread/openthread/blob/master/src/cli/README.md#eidcache) CLI command?

mwilniewiec, 2020-03-27:
From what I see our nodes have only 1 entry in the eidcache which is reasonable:
```
eidcache
fd3c:9b17:1634:0:fd33:96eb:becc:8097 201d
Done
```
This is ML-EID of our NCP and the only EID used by our application.

NCP on the other hand has lot of entries in Thread:AddressCacheTable property and it acts a bit strange, Im currently investigating it.
Sometimes it returns ~100 entries, sometimes it freezes for a few seconds and returns Thread:AddressCacheTable: <empty> (7)
It seems that the table is dropped too many times on our NCP.

jwhui, 2020-03-28:
> From what I see our nodes have only 1 entry in the eidcache which is reasonable:
> 
> ```
> eidcache
> fd3c:9b17:1634:0:fd33:96eb:becc:8097 201d
> Done
> ```
> 
> This is ML-EID of our NCP and the only EID used by our application.

Does the RLOC16 match what you expect on the NCP?

> NCP on the other hand has lot of entries in Thread:AddressCacheTable property and it acts a bit strange, Im currently investigating it.
> Sometimes it returns ~100 entries, sometimes it freezes for a [...]

The EID-to-RLOC cache gets cleared any time the NCP resets or reattaches to the network. Are you able to enable NCP logs and provide the log output from wpantund?

mwilniewiec, 2020-03-30:
The problem occurred one more time so I was able to do more investigation.
For a failed node we have:
1. empty eidcache -  (other nodes have valid NCP EID->RLOC entry)
1. pings to NCP RLOC work ~ 80% of the time
1. pings to NCP ML-EID are not working as described in the bug

The most important part I found and I think is the clue is the buffer info:
```
bufferinfo
total: 128
free: 2
6lo send: 0 0
6lo reas: 0 0
ip6: 0 0
mpl: 0 0
mle: 0 0
arp: 0 0
coap: 0 0
coap secure: 0 0
application coap: 63 126
Done
```
Does it mean that our application coap filled all the buffers and there is no more room for a/aq ?
I will surely have to look deeper into that problem.

mwilniewiec, 2020-03-30:
From what I see a/aq is a NON_CONFIRMABLE message so full buffers shouldn't be a problem in that case. I have a new feeling:
From what I see there is a 15 sec retry with 2* backoff on address resolver with max retry delay = 8 hours.
There could be a situation where in high packet loss environment this backoff is so big that we are waiting several minutes / hours to send next a/aq. Sniffer as described previously wouldn't show any ICMP as EID is not known and isn't showing a/aq as we are waiting for the next timer which can take half an hour.
Does is make sense? Do you know any easy way to confirm that? 
I can see that most of our problems lead us to the same thing: losing lots of packets and losing links between nodes.

PS. I also checked full buffers problem on such node and it's another mystery. This node wasn't sending app coap messages for at least 3 minutes. After that time send confirmable coap failed with No buffers error. We have default timeouts for coap so queue should be empty after maximum 93 seconds but it wasn't.

jwhui, 2020-03-30:
> From what I see a/aq is a NON_CONFIRMABLE message so full buffers shouldn't be a problem in that case. I have a new feeling:
> From what I see there is a 15 sec retry with 2* backoff on address [...]

Yes, we have seen this in practice. In Google Nest products, we actually configure `ADDRESS_QUERY_MAX_RETRY_DELAY` to 120 seconds (overriding the 8 hours stated in the Thread Specification). See similar discussion in https://github.com/openthread/openthread/issues/4542#issuecomment-584758632.

mwilniewiec, 2020-04-01:
I changed 8 hours to 240 sec in our project. I was considering your 120 suggestion but as we have more than 120 nodes in a single network I would prefer to avoid multicast every second. 
At the end we plan to remove (nodes->ncp) ML-EID communication and use service ALOCs instead as no extra multicasts are needed but we still have problems with invalid servers in network data. #4768 reported yesterday.

I have a feeling that the original problem is much deeper but reported issue seems to be an expected behavior, closing for now.

closed 2020-04-01 by mwilniewiec