openthread/openthread #4745: IAR can't compile two files with the same name (leader.cpp)

piotrkoziar, 2020-03-27:
**Describe the bug**
After recent changes (#4430), IAR build of OpenThread libraries fails with error like this:
```
Build error: Multiple tools write to the same file.
  File: foo\iar\_build\leader.o
  Tool 1: C/C++ Compiler, when processing foo\openthread\src\core\backbone_router\leader.cpp
  Tool 2: C/C++ Compiler, when processing foo\openthread\src\core\meshcop\leader.cpp
```
It is caused by two files with the same name (leader.cpp) that are processed during the compilation of SOURCES_COMMON in src/core/Makefile.am.

**To Reproduce**
Information to reproduce the behavior:
1. Git commit: #4430 or later
2. Use IAR toolchain to compile sources

**Solution**
I believe the best solution is to change filename of at least one leader.cpp file, i.e.: 
meshcop/~~leader.cpp~~ -> meshcop/**meshcop_leader.cpp**
meshcop/~~leader.hpp~~ -> meshcop/**meshcop_leader.hpp**

jwhui, 2020-03-27:
@piotrkoziar, thank you for raising this issue.

I agree that we should avoid having source files with duplicate filenames. Would you like to submit a PR with the proposed fix?

piotrkoziar, 2020-03-27:
@jwhui Thank you for the answer.

Yes, I will prepare PR with the fix.

jwhui, 2020-03-27:
> Yes, I will prepare PR with the fix.

@piotrkoziar, thanks!

closed 2020-03-30 by jwhui

OlivierGre, 2021-11-26:
@piotrkoziar How do you proceed to build the project with IAR?

piotrkoziar, 2021-12-04:
@OlivierGre Sorry but I no longer build OpenThread code with IAR, for more than a year. However, I was able to build the project at the time this issue was fixed. 
For me, a lot of work would be required to proceed with IAR compilation, especially after C++11 features were introduced to the code.

OlivierGre, 2021-12-06:
Hi Piotr. Ok thank you for your message.