openthread/openthread #4752: Getting OT_RADIO_RSSI_INVALID value quite frequently

NMzen, 2020-03-29:
Hi 

We are using nrfSDK and creating a thread network with nRF52840 devices. I am using otThreadGetParentAverageRssi to get the RSSI value. What I observed is that this function is returning OT_RADIO_RSSI_INVALID for quite a long time and then suddenly the value changes to -35 to -65 dBm on its own. Then again for sometime the value is RSSI_INVALID.

Can you describe what could be the reason for it and what can be done so that rssi value is a good one in the network.

Thanks

LuDuda, 2020-03-29:
Hello @NMzen the issue was recently fixed, and will be included in the next SDK release soon. Do you use current OT master branch?

Reference: #4537

NMzen, 2020-03-29:
umm.. No I am afraid not. I am using old nRF SDK.
Can you explain what was the problem? Because I don't see that in my local setup. However we are seeing it at one of the customer site.

LuDuda, 2020-03-29:
Which version of SDK are you using?

NMzen, 2020-03-29:
nRF5 SDK for Thread and Zigbee v2.0.0

NMzen, 2020-03-31:
@LuDuda your inputs awaited.

LuDuda, 2020-04-11:
@NMzen actually the `otThreadGetParentAverageRssi` function does not use the fast measurement of RSSI (`otPlatRadioGetRssi`) which I referred to before (so please ignore it - it was only related to Channel Monitor functionality).

The average RSSI for parent is calculated from the RSSI measurement on the received frame. I haven't seen such an error as well before.

Looking at the OT commit (SDK 2.0): 704511c96e0d093139e4b80ef0739ed2d701afb1

The `OT_RADIO_RSSI_INVALID` value can be returned only with the error code `OT_ERROR_FAILED`. Also the `RssAverager` module does not accept `OT_RADIO_RSSI_INVALID` value for calculation.

Saying that I think the problem might be that you call the `otThreadGetParentAverageRssi` function while having no active parent at the moment (`DETACHED` state)

jwhui, 2020-04-16:
@NMzen, any thoughts relative to https://github.com/openthread/openthread/issues/4752#issuecomment-612413225 above?

NMzen, 2020-04-17:
Ahh! This could be the case. Gimme a week's time to get it checked. I shall reply back to you.

Thank you team for all the help and guidance!! Regards.

NMzen, 2020-04-21:
Hi LuDa

I checked with my team. They told me that the parent was attached to the device as they were seeing the 127 value at the cloud only. Also they told me that only when they rebooted the device then the 127 value was gone and they started getting teh correct value. So i believe that No active parent case is not possible.

~ NM

jwhui, 2020-06-05:
@NMzen , could it be that the device was a router when it reported the value? In this case, the device would not have a parent.

jwhui, 2020-06-30:
Closing stale issue.

closed 2020-06-30 by jwhui