openthread/openthread #4768: Packets sent to Service ALOC not reaching destination (Part 2)

mwilniewiec, 2020-03-31:
This is a part 2 for #4472 as I found another easily reproducible path.
Reproduced that on POSIX from master branch.

1. Start Node L - a leader
1. Start Node C1 - a child (routereligible disable)
1. Start Node C2 - a child (routereligible disable)
1. On both C1 and C2 start a service:
   ```
   service add 44970 111 333
   netdataregister
   ```
1. Exit both C1 and C2 so that leader is left on his own
1. In leader network data we can see two servers: `0d05d4013333330 d05d402333333`
1. Wait for children timeout
1. When first child is timeouted we can see that its server was properly removed from netdata:
   ```
   > child table
   | ID  | RLOC16 | Timeout    | Age        | LQ In | C_VN |R|S|D|N| Extended MAC     |
   +-----+--------+------------+------------+-------+------+-+-+-+-+------------------+
   |   2 | 0xd402 |        240 |        237 |     3 |  204 |1|1|1|1| 3edfdec8d6b87619 |

   > netdatashow
   08040b0246050b0c80033131310d05d402333333
   ```
1. When second child is timeouted nothing happens and netdata remains invalid:
   ```
   > child table
   | ID  | RLOC16 | Timeout    | Age        | LQ In | C_VN |R|S|D|N| Extended MAC     |
   +-----+--------+------------+------------+-------+------+-+-+-+-+------------------+
   Done
   > netdatashow
   08040b0246050b0c80033131310d05d402333333
   ```
1. Invalid network data entry is persistent from now on
1. With more child services more invalid entries stay in the network data

jwhui, 2020-04-01:
@mwilniewiec, thank you for raising this issue.

I just tried this on latest master (7fd9902c38abfa88a8b237e439da9840f2a2150b), but was unable to reproduce.

What I did:
1. Start L:
```
```

mwilniewiec, 2020-04-01:
It was https://github.com/openthread/openthread/commit/bb86ae4c76b3de0ced16caf96e27c15e429d491c
Did you exit both C1 and C2 at the same time or did you wait for C1 to timeout before exiting C2?
I was exiting them at the same time so that their timeouts are very close.

jwhui, 2020-04-01:
@mwilniewiec, thanks for the clarification!

I can successfully reproduce the issue. Indeed, this issue occurs when two or more children timeout within a few seconds of each other. The issue is that we have a rate limiter on sending `SVR_DATA.ntf` messages (as required by the Thread Specification). If we fail to send a message due to the rate limiter, we will not try again.

I'll have a proposed fix posted shortly.

jwhui, 2020-04-01:
@mwilniewiec, I just submitted #4779. Please review.

closed 2020-04-02 by jwhui