openthread/openthread #4769: Make it possible for otPlatReset to be a callback

rob-the-dude, 2020-04-01:
**Is your feature request related to a problem? Please describe.**

On NetBSD, I'm using "crunch" to build an image (it's similar to Linux's "busy box") that effectively rebuilds all programs as a single executable.  To do this, it generally uses libraries as-is (the openthread libraries work fine), but it runs each executable's symbols through a script that tweaks the names to prevent conflicts.  Unfortunately, the openthread libraries link against a "otPlatReset" which is built into the executable -- but because the executable's symbols have been tweaked, the link cannot happen.

**Describe the solution you'd like**
I have a patch that allows you to OPTIONALLY (at compile time, via a new switch) build the libraries and executables in a way that work.  With this patch, "otPlatReset" is implemented by the library but it calls a function pointer that must be provided via a new API, "otSetPlatResetCallback".  The executable calls that API at startup and everything works fine.  It does add some code space and a variable for the function pointer, but I'm only using it on POSIX where that's OK for my application (and again, the whole feature is optional).

**Describe alternatives you've considered**
Haven't really tried any.

**Additional context**

jwhui, 2020-04-01:
@rob-the-dude, thanks for raising this issue.

I am generally OK with the approach. How about defining `otPlatReset()` somewhere in the platform driver layer (i.e. [src/posix/platform](https://github.com/openthread/openthread/tree/master/src/posix/platform)) and defining an API specific to the posix platform for use in [src/posix/main.c](https://github.com/openthread/openthread/blob/master/src/posix/main.c). In this way, we don't need to make any changes to the existing `otPlat*` APIs. Thoughts?

bukepo, 2020-04-02:
Hi @rob-the-dude , I don't quite understand the problem yet.  do you mean openthread libraries are dynamically linked? If so can static link solve this issue?

rob-the-dude, 2020-04-02:
Hi, @bukepo.  No, static linking doesn't solve the issue -- everything in crunch is statically linked.

Here's how crunch works:

#1) First, you build all the libraries you want in your system.  All of the openthread libraries that run on the host are built during that stage, and they work more or less as you would expect them in non-crunch builds.

#2) Next, you COMPILE the sources for every executable in the system (this is where "ot-cli" or "ot-daemon" get built).  But before you link, you run all the object files through a name-mangling process that pre-pends the executable's name onto the symbol.  This is necessary, because many executables have functions with the same name (indeed, ALL of them have at least one conflict: "main").

#3) Now crunch creates a new "main" function that looks at "argv[0]" to see how the executable was ACTUALLY called, and then call the actual correct executable routine.  So if the program was run as "/bin/true", it calls "true_main" -- if the program was called "ot-daemon", it calls "otdaemon_main".  (It's basically a very long if-strcmp-else-if-strcmp-else-if ... function.)

#4) Now link the binary, and call it something like "crunch".  (The main benefit is that everything in any common library and executable that is not used by something gets dead-stripped for the entire system, resulting in file system space savings.  It has an added benefit for the VM system, in that every executable shares the same read-only code pages, resulting in some RAM savings.)

#5) Finally, create a file system that hard-links all the executables you built:

     ln crunch /bin/true
     ln crunch /sbin/ot-daemon
     ... etc ...

Now you can run any of those programs, but it's actually running the same executable each time.

So now, getting back to this SPECIFIC problem, the openthread libraries (built in #1) have a dependency on the function "otPlatReset".  But in the current code, that function is implemented in the application.  And in step #2 above, it got "mangled" to "otdaemon_otPlatReset".  And so you get a link error.

(FWIW, I think dynamic linking could have SIMILAR but not the same issues.  However, I could imagine a dynamic linker that allows applications to export functions that are imported by loaded dynamic libraries -- though I don't know if that actually happens in practice.)

The problem here is that the function has been renamed because of where it's implemented.  To fix it, we need to implement it in the library, not the application.  But openthread's architecture really wants the application to handle this logic (makes perfect sense).  The "solution" I've proposed to the problem is to have a callback from the library to the application, along with an additional API to set the callback.  (Such a solution would ALSO solve dynamic linking issues, if they exist.)

bukepo, 2020-04-02:
Thank you for the detailed explanation and proposing a solution. I was wondering if there were ways to keep certain symbols. From https://www.freebsd.org/cgi/man.cgi?query=crunchgen&sektion=1&n=1, it seems possible, there is a **keep** option for programs to avoid certain symbols being renamed. Can that solve the problem?

rob-the-dude, 2020-04-03:
Ah, I had intended to go look and see if crunch had something like that, but forgot to look before trying to get some of my openthread patches cleared out.

NetBSD has a "keep symbols" option, and using that seems to allow otPlatReset to work.  So I can use that to work around this issue.  That being said, this still seems like a limitation in OpenThread, where the platform needs to have a work-around.  But I personally won't be blocked.

Should I abandon the PR, and leave OpenThread's limitations intact?

bukepo, 2020-04-03:
Hi @rob-the-dude, I think changing objects' symbol names is an exceptional case, so I tend not to handle it in OpenThread.

rob-the-dude, 2020-04-03:
I agree with that.

But having libraries link against application-exported functions isn't very clean, either.  The reason crunch works with most applications (indeed, this is the first one I've encountered) is because they avoid that kind of misdirection.  Is this really a Good Thing(tm) that OpenThread does?

bukepo, 2020-04-03:
Hi @rob-the-dude , we have several application-exported functions except otPlatReset(), such as otPlatLog(), otTaskletsSignalPending(), otPlatCAlloc() and otPlatFree(). See:

* https://github.com/openthread/openthread/blob/master/examples/apps/cli/main.c
* https://github.com/openthread/openthread/blob/master/examples/apps/ncp/main.c

I think this is because how these platform APIs should be implemented can only be decided at the application layer of the current design. I agree to add a callback for now.

rob-the-dude, 2020-04-25:
Given those other APIs, and the feature in crunchgen that unblocks my own work (thanks!), let's defer this until all of those APIs can be addressed -- which doesn't seem like a high priority.

closed 2020-04-25 by rob-the-dude