openthread/openthread #4770: SOCK_NONBLOCK (passed to SocketWithCloseExec in posix netif) is Linux-specific

rob-the-dude, 2020-04-01:
**Is your feature request related to a problem? Please describe.**

SOCK_NONBLOCK is a Linux-specific feature.  Some BSDs, including macOS, don't have it.

**Describe the solution you'd like**

It looks like the posix code is trying to wrap the socket calls with the SocketWithCloseExec API (likely because SOCK_CLOEXEC is also non-portable).  Perhaps a new bool parameter indicating that the socket should be marked non-blocking?

**Describe alternatives you've considered**
I can also #ifdef around the issue and make the socket non-blocking in the code, but it just looks uglier.

gjc13, 2020-04-03:
Would adding `fcntl(fd, O_NONBLOCK)` suffice for  BSD?

rob-the-dude, 2020-04-03:
Yes, that's what is needed.

However, I was looking at this again.  Why do we mark it non-blocking?  We never call "read/recv" on it unless "select" has indicated it's ready for reading.  And we never write to the socket, so we will never stall for such a case.

We don't mark the other sockets/descriptors as non-blocking.  In particular, the AF_NETLINK socket (sNetlinkFd) is used almost exactly the same way as the MLD socket (we poll for readability on select, and then read).  Why is one non-blocking and the other blocking?  The IP ioctl socket (sIpFd) is left in blocking mode, but all we use it for is ioctl, so that seems right.  And the tunnel descriptor (sTunFd) is also in blocking mode -- that may be intended (the main consequence is that "processReceive" may block in "write" while sending packets to the kernel, though in practice that's likely extremely unlikely -- we do not handle the EWOULDBLOCK error (or any error) from "write" particularly gracefully, but we don't fail -- the packet is simply lost).

Personally, it seems the easiest thing to do may simply be to remove SOCK_NONBLOCK and don't worry about.  I don't think the socket needs to be non-blocking for the use case we have here. But I'm willing to be to be educated why that's wrong, if I can also be educated why the other sockets are properly blocking.

gjc13, 2020-04-07:
https://stackoverflow.com/questions/5351994/will-read-ever-block-after-select

Chances are that the read after select will block so we add NONBLOCK option to be safe.

gjc13, 2020-04-07:
I found some of the fds in netif.cpp not opened with NONBLOCK option. I'll also update these fds.

closed 2020-04-21 by jwhui