openthread/openthread #4785: Routers unable to boot

NMzen, 2020-04-03:
We are using raytech dongles (based on nrf52840 chipset) as the routers for our thread network. In one of our site, we saw that randomly some of the routers stopped communicating and when we checked them physically, saw that they were not booting. This we are seeing only on one of our sites and this had big setup of routers and SEDs.

On further analysis we got to know that the routers are in continuous reboot mode. We have initialized the watchdog timer in our application and somehow the code is stuck somewhere in otInstanceInitSingle() function which is causing the reboot. 

To make those routers work again, we just erase the complete flash and reprogram them with the same code and it works fine. I believe that there is something related to the datasets that gets saved on the flash. Note that our application is not writing anything on the nordic flash.

Any idea what is the reason for this flash corruption. 

FYI - We are using nrfSDK for Thread and Zigbee 2.0.0 for building the application.

NM

jwhui, 2020-04-03:
Maybe you are running into the same issue as #4641 ?

jwhui, 2020-04-16:
@NMzen, do the fixes in #4234 and/or #4640 resolve your issue?

NMzen, 2020-04-17:
Hi @jwhui , we couldn't test it, as it is on a client site. But we believe what you have explained could be the reason for what we are seeing. We have changed our OT stack as per the suggestion. And plan is to test next month on the site.

jwhui, 2020-04-17:
> Hi @jwhui , we couldn't test it, as it is on a client site. But we believe what you have explained could be the reason for what we are seeing. We have changed our OT stack as per the suggestion. And plan is to test next month on the site.

@NMzen, thanks for the update. I'm going to close this issue for now. Feel free to reopen if you feel this issue has not been resolved.

closed 2020-04-17 by jwhui