openthread/openthread #4796: [build] Unaligned load/store compiler warning

dismirlian, 2020-04-06:
Hi all, GCC/Clang provide the -Wcast-align warning switch, which tells the compiler to emit warnings whenever a pointer is cast such that the required alignment of the target is increased. For instance, `*reinterpret_cast<uint32_t *>(GetEnterpriseNumberLocation())` would emit a warning.

This issue is related to #4617.

In the former example, the compiler could emit instructions which access unaligned memory. Cortex M4 mostly supports unaligned accesses, but store/load double/multiple instructions (LDM, STM, LDRD and STRD) will always generate a usage fault. In general, I guess that the compiler doesn't generate these instructions frequently, so the code works. But I think we should enable the warning and fix the potential issues because the code would break with a new compiler version or different optimization levels.

I enabled -Wcast-align, and I found these instances of possible problems in (core) code.

In `heap.hpp`:
https://github.com/openthread/openthread/blob/c6a258e3a5bd90aa26d8245c435c0ae5797027f4/src/core/utils/heap.hpp#L94-L97

https://github.com/openthread/openthread/blob/c6a258e3a5bd90aa26d8245c435c0ae5797027f4/src/core/utils/heap.hpp#L107-L110

These are not really issues, because `CAlloc` always aligns `size`. In this case, there are two ways to tell the compiler that this isn't a problem:
1) Use GCC/Clang `__builtin_assume_aligned`:
```
uint16_t GetNext(void) const
{
    const uint8_t *ptr = reinterpret_cast<const uint8_t *>(this) + sizeof(mSize) + mSize;
    return *reinterpret_cast<const uint16_t *>(__builtin_assume_aligned(ptr, 1));
}
```
(of course, wrapped in a `OT_TOOL...` macro). This alternative doesn't have any downside, except possibly for compiler support.

2) Something like this:
```
uint16_t GetNext(void) const
{
    const uint8_t *ptr = ;
    return *(reinterpret_cast<const uint16_t *>(this) + sizeof(mSize) / 2 + mSize / 2);
}
```
This alternative increases the code size by 16 bytes on my test platform.

In `network_data_tlvs.hpp`:

https://github.com/openthread/openthread/blob/c6a258e3a5bd90aa26d8245c435c0ae5797027f4/src/core/thread/network_data_tlvs.hpp#L927-L931

https://github.com/openthread/openthread/blob/c6a258e3a5bd90aa26d8245c435c0ae5797027f4/src/core/thread/network_data_tlvs.hpp#L956-L962

These are actually potential issues (there is even a comment that flags them!). Two potential solutions:

1) Change the code to `TlvUint32` helper class. This seems the "correct" thing to do, but I think it is a rather large change.

2) Something like this:
```
In toolchain.h:
typedef uint32_t packed_uint32_t __attribute__((aligned(1)));   //(or equivalent for other compilers).

In network_data_tlvs.hpp:
return HostSwap32(*reinterpret_cast<const packed_uint32_t *>(GetEnterpriseNumberLocation()));
```

What do you think?

jwhui, 2020-04-07:
@dismirlian, thank you for raising this issue!

For the heap code, I would tend to prefer the second option because it avoids compiler support. 16 byte code overhead seems OK for this.

For the network data code, how about using:
https://github.com/openthread/openthread/blob/c6a258e3a5bd90aa26d8245c435c0ae5797027f4/src/core/common/encoding.hpp#L145-L149

dismirlian, 2020-04-07:
Hi @jwhui, thanks for your suggestions!

> 1. For the heap code, I would tend to prefer the second option because it avoids compiler support. 16 byte code overhead seems OK for this.

Actually, a better alternative would be an intermediate cast to a `void *`; that will get rid of the warning without any impact to code size.

> 2. For the network data code, how about using:
>    https://github.com/openthread/openthread/blob/c6a258e3a5bd90aa26d8245c435c0ae5797027f4/src/core/common/encoding.hpp#L145-L149

Great.

jwhui, 2020-04-07:
@dismirlian, sounds good. Are you planning to submit a PR?

dismirlian, 2020-04-07:
@jwhui yes, but first I need to figure out what to do with the platforms. I can enable the warning for the core code only, for now. What do you think?

jwhui, 2020-04-07:
Enabling the warning for code in `src/` is a good start! Thanks, @dismirlian.

jwhui, 2020-04-30:
@dismirlian , have you been able to make any progress on this?

dismirlian, 2020-04-30:
@jwhui, I'll have to re-analyze this because of the changes in #4806; maybe it's not relevant any more. I'll try to do it over the weekend.

closed 2020-05-07 by jwhui