openthread/openthread #4799: [ci] apt-get install fails too often in github actions

gjc13, 2020-04-07:
**Describe the bug**
The `apt-get update` command fails frequently in github actions, forcing the developers to manually re-trigger the jobs.

**To Reproduce**
Re-run the github actions.

**Expected behavior**
The bootstrap stage should always succeed.

**Console/log output**
```
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/main/g/glibc/libc6-dev-i386_2.27-3ubuntu1_amd64.deb  Could not connect to azure.archive.ubuntu.com:80 (52.177.174.250), connection timed out
Fetched 43.9 MB in 32s (1362 kB/s)
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/main/g/glibc/libc6-x32_2.27-3ubuntu1_amd64.deb  Unable to connect to azure.archive.ubuntu.com:http:
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/main/g/glibc/libc6-dev-x32_2.27-3ubuntu1_amd64.deb  Unable to connect to azure.archive.ubuntu.com:http:
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/main/p/python-setuptools/python3-setuptools_39.0.1-2_all.deb  Unable to connect to azure.archive.ubuntu.com:http:
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

jwhui, 2020-04-07:
This is a known issue in GitHub Actions: https://github.com/actions/virtual-environments/issues/675

closed 2020-04-07 by jwhui

jwhui, 2020-04-08:
FYI https://github.com/actions/virtual-environments/issues/675#issuecomment-610712578: 

> We've engaged the mirror hosts and they have added capacity to improve it's reliability. I'll keep this open until we can confirm the issue is resolved. Please let us know if this continues to occur. Thank you for the reports.

stryker-sk, 2024-03-01:
Getting this a bunch today now