openthread/openthread #4807: [joiner] tear down of DTLS session

LuDuda, 2020-04-09:
Currently, the DTLS session is closed by the Joiner after getting the [Joiner Finalize Response](https://github.com/openthread/openthread/blob/master/src/core/meshcop/joiner.cpp#L510).

According to the specification, because OpenThread does not implement Joiner Provisioning, the Joiner Session must be closed immediately after the Joiner Entrust message (which comes later).

Is there any specific reason for the current behavior?

jwhui, 2020-04-09:
Thread 1.1.1 Section 8.4.5.2 states the following:

> The Commissioner is provided with a mechanism to securely exchange subsequent
> application-specific provisioning data to the new node.
> 
> If the Joiner appealed for a specific Commissioning application and the application matched,
> then after the Joiner Entrust, the Joiner Session remains open for vendor-specific
> provisioning, and is only closed when the vendor-specific provisioning protocol specifies.
> 
> In all other cases, **the Joiner Session is closed immediately after Joiner Finalization**.

I believe the current Joiner behavior is implemented based on the above text.

Can you point to where it specifies that the Joiner Session is closed immediately after the Joiner Entrust message?

LuDuda, 2020-04-09:
Thank you @jwhui. I opened some of the Thread Specification 1.1.0 by accident where the text is exactly the same however **Joiner Finalization** is replaced with **Joiner Entrust**.

closed 2020-04-09 by LuDuda