openthread/openthread #4810: Modifying mlallowzerohoplimit = true causes the COAP data to be unable to be sent

Feng573795502, 2020-04-10:
Modifying mlallowzerohoplimit = true causes the COAP data to be unable to be sent.

I am implementing similar airpods near the networking. Other functions have been implemented but as long as HopLimit is set to 0, the coap message cannot be sent and the server cannot receive it.

1. Git commit id 00f8567e88fffb93b88673350399d330aa9da4e6
2. NRF52840 4.00
`void thread_coap_joiner_request_send(void)
{
    otError       error = OT_ERROR_NO_BUFS;
    otMessage   * p_request;
    otMessageInfo message_info;
    otInstance  * p_instance = thread_ot_instance_get();

    do
    {
        p_request = otCoapNewMessage(p_instance, NULL);
        if (p_request == NULL)
        {
            return;
        }

        otCoapMessageInit(p_request, OT_COAP_TYPE_NON_CONFIRMABLE, OT_COAP_CODE_GET);
        otCoapMessageGenerateToken(p_request, 2);

        error = otCoapMessageAppendUriPathOptions(p_request, JOIN_PATH);
        ASSERT(error == OT_ERROR_NONE);

        //POLL Set
        uint32_t err_code = thread_poll_period_response_set();
        if (err_code == NRF_ERROR_BUSY)
        {
            return;
        }

        memset(&message_info, 0, sizeof(message_info));
        message_info.mPeerPort    = OT_DEFAULT_COAP_PORT;
        memcpy(&message_info.mPeerAddr, &m_operat.peer_address, sizeof(message_info.mPeerAddr));
				
        //message_info.mAllowZeroHopLimit = true;

        error = otCoapSendRequest(
            p_instance, p_request, &message_info, thread_coap_joiner_request_handler, p_instance);

    } while (false);

    if (error != OT_ERROR_NONE && p_request != NULL)
    {
        otMessageFree(p_request);
    }
}`

jwhui, 2020-04-10:
@Feng573795502, if you are setting [otMessageInfo::mAllowZeroHopLimit](https://openthread.io/reference/struct/ot-message-info#mallowzerohoplimit) to true, then an IPv6 Hop Limit of zero will result in the packet being dropped locally on the device and not transmitted over the radio.

https://github.com/openthread/openthread/blob/2f618bd4a72501a2f31e23d50a4f388b9a8c6b9b/src/core/net/ip6.cpp#L1261-L1265

Feng573795502, 2020-04-11:
> @Feng573795502, if you are setting [otMessageInfo::mAllowZeroHopLimit](https://openthread.io/reference/struct/ot-message-info#mallowzerohoplimit) to true, then an IPv6 Hop Limit of zero will result [...]

`otCoapMessageInit(p_request, OT_COAP_TYPE_CONFIRMABLE, OT_COAP_CODE_GET);`

Modify the communication like this

jwhui, 2020-04-12:
RFC 7252 Section 8.1 states:

>    A multicast request is characterized by being transported in a CoAP
>    message that is addressed to an IP multicast address instead of a
>    CoAP endpoint.  **Such multicast requests MUST be Non-confirmable.**

jwhui, 2020-04-16:
Closing stale issue.

closed 2020-04-16 by jwhui