openthread/openthread #4813: Build failing for microchip's SAMR21 using ASF.

thk3695, 2020-04-11:
Hello guys!
I found the following bug while building openthread drivers for SAMR21.

**Description**
I am trying to build openthread drivers for Microchip's SAMR21_XPLAINED_PRO board. I followed the build instructions mentioned in the microchip platform folder:
[Microchip_SAMR21](https://github.com/openthread/openthread/tree/master/examples/platforms/samr21)

I encountered the following errors during the build:
```
Makefile:1283: recipe for target '../../../third_party/microchip/asf/thirdparty/wireless/avr2130_lwmesh/source/phy/at86rf233/src/libopenthread_samr21_a-phy.o' failed
```

**To Reproduce**
After setting up the environment, and running,
```
make -f examples/Makefile-samr21
```
gives me the following error:
**Console/log output:**
```
In file included from /home/thk3695/git/openthread/examples/../third_party/microchip/asf/thirdparty/CMSIS/Include/cmsis_compiler.h:48:0,
                 from /home/thk3695/git/openthread/examples/../third_party/microchip/asf/thirdparty/CMSIS/Include/core_cm0plus.h:115,
                 from /home/thk3695/git/openthread/examples/../third_party/microchip/asf/sam0/utils/cmsis/samr21/include/samr21g18a.h:223,
                 from /home/thk3695/git/openthread/examples/../third_party/microchip/asf/sam0/utils/cmsis/samr21/include/samr21.h:55,
                 from /home/thk3695/git/openthread/examples/../third_party/microchip/asf/sam0/utils/header_files/io.h:56,
                 from /home/thk3695/git/openthread/examples/../third_party/microchip/include/compiler.h:57,
                 from /home/thk3695/git/openthread/examples/../third_party/microchip/asf/thirdparty/wireless/services/trx_access/trx_access.h:58,
                 from /home/thk3695/git/openthread/examples/../examples/platforms/samr21/../../../third_party/microchip/asf/thirdparty/wireless/avr2130_lwmesh/source/phy/at86rf233/src/phy.c:41:
/home/thk3695/git/openthread/examples/../examples/platforms/samr21/../../../third_party/microchip/asf/thirdparty/wireless/avr2130_lwmesh/source/phy/at86rf233/src/phy.c: In function 'PHY_DataReq':
/home/thk3695/git/openthread/examples/../third_party/microchip/asf/thirdparty/CMSIS/Include/cmsis_gcc.h:41:50: error: expected expression before '__asm'
   #define __ASM                                  __asm
                                                  ^
```

**Additional Info:**
OS: Linux Ubuntu 18.04 LTS 
Linux Version: 5.3.0-46-generic
ASF version: 3.48
arm-none-eabi-gcc version: 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

jwhui, 2020-04-11:
@thk3695, please try using [Advanced Software Framework v3.45](http://ww1.microchip.com/downloads/en/DeviceDoc/asf-standalone-archive-3.45.0.85.zip) instead.

thk3695, 2020-04-12:
Thank you @jwhui , using ASF 3.45 helped.

closed 2020-04-12 by thk3695