openthread/openthread #4814: [config] Why there are two user-provide config files?

wgtdkp, 2020-04-12:
**Is your feature request related to a problem? Please describe.**
We have `OPENTHREAD_CONFIG_FILE` and `OPENTHREAD_PROJECT_CORE_CONFIG_FILE` to enable allow users select features:
1. why would we need two different config files?
2. `OPENTHREAD_CONFIG_FILE` is exported in `<openthread/config.h>`, why is exporting this necessary? since this is provided by user, there seems no need to export it to user in reverse.

**Describe the solution you'd like**
If possible, use one single config file and keep it away from public headers.

**Describe alternatives you've considered**

**Additional context**

wgtdkp, 2020-04-12:
@abtink @bukepo

jwhui, 2020-04-12:
@wgtdkp, the purpose of `<openthread/config.h>` is to pull in the header file typically autogenerated by `configure.ac`. It includes information about what header files are provided by the system and macros for the package information (e.g. package name, package string, etc.). The reason why `OPENTHREAD_CONFIG_FILE` exists is to support build systems that do not include `configure.ac`.

Given that OpenThread's public headers do not rely on these macros, we could avoid exporting them in most cases. The few exceptions may be package metadata macros, which may be valuable to some downstream projects.

wgtdkp, 2020-04-12:
@jwhui Thanks for explanation, it is clear.

closed 2020-04-12 by wgtdkp