openthread/openthread #4827: REED unable to synchronize network data to Leader when Leader has smaller `routerupgradethreshold`

simonlingoogle, 2020-04-14:
**Describe the bug**
If the Leader has smaller `routerupgradethreshold` than the REED, the REED will keep sending `AddressSolicit` and being rejected by the Leader. In the meantime, the REED won't be able to synchronize local network data to Leader, e.x. it won't be able to become Primary BBR. 

**To Reproduce**

1. Git commit id: master
2. IEEE 802.15.4 hardware platform: simulation
3. Build steps: 
```
make -f examples/Makefile-simulation BACKBONE_ROUTER=1 THREAD_VERSION=1.2
```
4. Network topology
```
Leader---REED
```
Execution steps:
  1. Start the leader and wait for it to become leader
  2. Set `routerupgradethreshold` to 1 on Leader: `routerupgradethreshold 1`
  3. Start the REED
  4. Start BBR service on REED: `bbr enable`
  5. Wait for a long time, but REED never become Primary BBR:
```
node 2> bbr enable
Done
node 2> 
node 2> bbr
BBR Primary: None
Done
```

**Expected behavior**
The REED should become Primary BBR. 

**Proposed fix**
A simple fix can be if the REED receives `a/as` ACK with `status == 1 (No Address Available)`, it can update its `routerupgradethreshold` like below:
```cpp
routerupgradethreshold = min(routerupgradethreshold, active_router_count)
``` 
This can help the node exit `REED` status and return back to child state. 

Thoughts? @librasungirl @jwhui

librasungirl, 2020-04-14:
It should be a common issue regardless of the Thread version. And it may happen in user scenarios when Thread devices from different vendors have different `routerupgradethreshold` configuration (though default 16, but it is configurable).

@simonlingoogle in your proposal, will it change the [mRouterUpgradeThreshold](https://github.com/openthread/openthread/blob/master/src/core/thread/mle_router.hpp#L827)?

I may suggest to introduce a variable like `mRunningRouterUpgradeThreshold`, thus when the device need to become Leader due to lost of Leader, it still have the chance to adopt its original `mRouterUpgradeThreshold` value when deciding whether or not to allocate Router ID for the others.

simonlingoogle, 2020-04-14:
> It should be a common issue regardless of the Thread version.

Yes, it is a common issue. I am using `BBR service` mainly for easy illustration. The issue can happen whenever the Leader has smaller `routerupgradethreshold` than the REED.

simonlingoogle, 2020-04-14:
> @simonlingoogle in your proposal, will it change the [mRouterUpgradeThreshold](https://github.com/openthread/openthread/blob/master/src/core/thread/mle_router.hpp#L827)?

Yes.

> I may suggest to introduce a variable like `mRunningRouterUpgradeThreshold`, thus when the device becomes Leader somehow e.g. due to lost of Leader, it still have the chance to adopt its original `mRouterUpgradeThreshold` value when deciding whether or not to allocate Router ID for the others.

I would prefer not introducing another variable a little bit because extra RAM and code will be required for storing, accessing and updating the new `mRunningRouterUpgradeThreshold` variable, whereas in the proposed fix, no other logic change is required after [mRouterUpgradeThreshold](https://github.com/openthread/openthread/blob/master/src/core/thread/mle_router.hpp#L827) is altered. 

We also need to clear `mRunningRouterUpgradeThreshold` when  [mRouterUpgradeThreshold](https://github.com/openthread/openthread/blob/master/src/core/thread/mle_router.hpp#L827) is configured again otherwise the new `mRouterUpgradeThreshold` is ineffective, to the surprise of users.

librasungirl, 2020-04-14:
If so, some scenarios you may need to consider
- how many active routers would you expect in the Thread Network formed by the device itself if it once joined  a Thread network with lower `upgradethreshold`?
- what if there is partition and migration, and the device attaches to a partition with a Leader which has bigger `upgradethreshold` setting, should it try to become active router? How to update to a bigger `upgradethreshold`?

simonlingoogle, 2020-04-14:
> * how many active routers would you expect in the Thread Network formed by the device itself if it once joined  a Thread network with lower `upgradethreshold`?

`mRouterUpgradeThreshold` will keep unchanged without further configuration. 

> * what if there is partition and migration, and the device attaches to a partition with a Leader which has bigger `upgradethreshold` setting, should it try to become active router? How to update to a bigger `upgradethreshold`?

It can't. But it would be the responsibility of the network administrator to set `routerupgradethreshold` configuration correctly in the whole network. So, the proposed fix is a simple workaround for OpenThread to "fix" the incorrect configuration.

simonlingoogle, 2020-04-15:
@jwhui Any thoughts on this issue?

jwhui, 2020-04-15:
> @jwhui Any thoughts on this issue?

The Thread Specification sets `ROUTER_UPGRADE_THRESHOLD` to 16 and does not provide any provisioning for changing the value. As a result, any Thread network in production should this value.

Is changing the `ROUTER_UPGRADE_THRESHOLD` needed for testing when operating as a reference device in the Thread Test Harness? If so, I would be fine introducing a new variable. We may also want to wrap the `ROUTER_UPGRADE_THRESHOLD` setting under the `REFERENCE_DEVICE` build feature.

librasungirl, 2020-04-15:
> The Thread Specification sets ROUTER_UPGRADE_THRESHOLD to 16 and does not provide any provisioning for changing the value. As a result, any Thread network in production should this value.

👍 
Confirmed there is note in the API as well.
https://github.com/openthread/openthread/blob/b37528645a57ca963dc847a67a2d783d401c20c6/include/openthread/thread_ftd.h#L342-L354

> Is changing the ROUTER_UPGRADE_THRESHOLD needed for testing when operating as a reference device in the Thread Test Harness? If so, I would be fine introducing a new variable. We may also want to wrap the ROUTER_UPGRADE_THRESHOLD setting under the REFERENCE_DEVICE build feature.

Yes,  different from Thread 1.1 TH, to avoid a large scale testbed for REED DUT role, Thread 1.2 TH chooses to update the `ROUTER_UPGRADE_THRESHOLD` on the reference leader device which would reject the address solicit from DUT to fake the test scenario.

simonlingoogle, 2020-04-15:
> Yes, different from Thread 1.1 TH, to avoid a large scale testbed for REED DUT role, Thread 1.2 TH chooses to update the `ROUTER_UPGRADE_THRESHOLD` on the reference leader device which would reject the address solicit from DUT to fake the test scenario.

Despite that, the Thread 1.2 certification do requires that the REED DUT, which is not necessarily a Reference Device, to work normally when the Leader keeps rejecting it's `a/as`. Unfortunately the current OT implementation does not work well in such condition. 
We would require the new variable `mRunningRouterUpgradeThreshold` logic on all REED devices (can not wrap with `REFERENCE_DEVICE`).

jwhui, 2020-04-15:
> Despite that, the Thread 1.2 certification do requires that the REED DUT, which is not necessarily a Reference Device, to work normally when the Leader keeps rejecting it's `a/as`. Unfortunately the [...]

I don't think we should be assuming a different value for the router upgrade threshold.

How about we allow sending `SVR_DATA.ntf` after a REED receives an `ADDR_SOL.rsp` message with Status TLV indicating "No Address Available"?

simonlingoogle, 2020-04-15:
> How about we allow sending `SVR_DATA.ntf` after a REED receives an `ADDR_SOL.rsp` message with Status TLV indicating "No Address Available"?

Sounds great to me. This will make OT work normally in REED state. I will create a PR to enhance accordingly.

simonlingoogle, 2020-04-15:
PR #4834 has been created for addressing this issue. 
However I found there is some issues in the current fix. 

Whenever the REED tries to send `SVR_DATA.ntf`, it will check `ExpectBecomeRouterSoon` and giveup sending if `ExpectBecomeRouterSoon` returns `TRUE`.
If the REED sends `SVR_DATA.ntf` before sending `ADDR_SOL.req`, and then the `ADDR_SOL.req` is rejected, the REED will have to wait for a period of time (at most `GetRouterSelectionJitterTimeout()+1` seconds) for the next `SVR_DATA.ntf`. 
Same issue may also apply to future Thread 1.2 DUA registraion features, in which DUA registration will be delayed until next renewing. 

So, should the REED try to resend `SVR_DATA.ntf` and `DUA registration` when its `a/as` is rejected?

closed 2020-04-18 by jwhui