openthread/openthread #4833: [travis] run unit tests for 1.2 features

Irving-cl, 2020-04-15:
Since we're moving forward to 1.2, shall we add a test entry in travis which runs unit tests with `OT_THREAD_VERSION_1_2`? I think currently all unit tests are run with `
OT_THREAD_VERSION_1_1`. So it's infeasible if we want to test some 1.2 features in unit tests now.

For example, I added some functions for parsing version 2015 frame in #4826 , but as the feature would only exist when 1.2 is enabled. So no unit tests for it are actually be run on travis now.

Shall we add an entry in travis for running 1.2 unit tests?

jwhui, 2020-04-15:
We should definitely be running unit tests with Thread 1.2 build feature enabled.

Since we are transitioning from Travis to GitHub Actions, we should be adding these tests to GitHub Actions instead.

For reference, we are running Thread 1.2 functional tests here:

https://github.com/openthread/openthread/blob/b37528645a57ca963dc847a67a2d783d401c20c6/.github/workflows/simulation.yml#L377-L401

closed 2020-04-23 by Irving-cl

simonlingoogle, 2020-04-23:
Why do we close this issue ? @Irving-cl

Irving-cl, 2020-04-24:
> Why do we close this issue ? @Irving-cl

I have added unit tests for Thread 1.2 in #4842

simonlingoogle, 2020-04-24:
> > Why do we close this issue ? @Irving-cl
> 
> I have added unit tests for Thread 1.2 in #4842

Thanks for clarification. I thought this issue was about running cert tests.