openthread/openthread #4835: [posix] examples in src/posix/README.md doesn't work

wgtdkp, 2020-04-15:
**Describe the bug**
The nrf52840 examples in src/posix/README.md stops working when start with command `./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli /dev/ttyACM0 115200`. The program exit silently with an exit code `6`, which is represented as `OT_EXIT_RADIO_SPINEL_NO_RESPONSE`.

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id: bfef9c6a035fe33e4727db6b85dd06750de74d51
2. IEEE 802.15.4 hardware platform: nrf52840 DK
3. Build steps: follow the examples in src/posix/README.md
4. Network topology: none.

**Expected behavior**
it is expected that the `ot-cli` will start successfully anf output a rpompt:
```shell
>
```

**Console/log output**
no output but has a non-zero exit value (6 = OT_EXIT_RADIO_SPINEL_NO_RESPONSE).

**Additional context**
The nrf52840 examples with `USB=0` can be fixed by change the command to:
```shell
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli /dev/ttyACM0 115200N1H
```
Because the nrf52xxx drivers enable hardware flow control by default:
https://github.com/openthread/openthread/blob/bfef9c6a035fe33e4727db6b85dd06750de74d51/examples/platforms/nrf528xx/nrf52840/transport-config.h#L77
So enable flow control with the `H` at the end of the command works.

**But the example with `SUB=1` keeps failing and the exit with the same value, when started with same command above.**

jwhui, 2020-06-30:
@wgtdkp , is this still an issue?

wgtdkp, 2020-07-01:
No, should be closed.

closed 2020-07-01 by wgtdkp