openthread/openthread #4863: Commissioning README results in failure

arnulfrupp, 2020-04-20:
**Describe the bug**
The “OpenThread CLI – Commissioning” example (/src/cli/README_COMMISSIONING.md) fails. 

**To Reproduce**
Following the the README guide step by step results in "in failed [NotFound]" after "joiner start J01NME" in Start Joiner / 1. Start the Joiner process:  > joiner start J01NME

**Expected behavior**
Either the README should include an additional step setting the panid to the value found in the network name after a scan or a well known panid should be used on both the commissioner or joiner.

jwhui, 2020-04-21:
@arnulfrupp, thank you for raising this issue!

I agree that the readme should add some clarification - this is an issue that has tripped up a few people already.

The issue is that the PAN ID needs to be set to 0xffff when you truly want to scan across all networks. The PAN ID is reset to 0xffff upon a factory reset, but there are cases where you want to perform commissioning again even without a factory reset. We can certainly update the readme to make this more clear.

arnulfrupp, 2020-04-21:
Good point. This issue will not arise if the joiner is factory fresh or reset.
Would be good to add a remark in the readme that the PAN ID must be set explicitly or with factory reset.

jwhui, 2020-04-21:
@arnulfrupp, I just submitted #4868 that adds a factory reset step to the beginning of the joiner process. I think this is the simplest approach that will avoid the issues raised. Please review.

PR #4868 (2020-04-22): [cli] add `factoryreset` for Joiner in README_COMMISSIONING.md
Resolves #4863
closed 2020-04-22 by jwhui