openthread/openthread #4869: OT_CONFIG and OT_PRIVATE_DEFINES specified too late to impact platform

rob-the-dude, 2020-04-21:
**Describe the bug**
In the top level CMakeLists.txt, if OT_CONFIG is defined, then it is added to OT_PRIVATE_DEFINES only AFTER the OT_PLATFORM directory has been added.  Similarly, any externally provided options (OT_PLATFORM_DEFINES) are also added to OT_PRIVATE_DEFINES only AFTER the OT_PLATFORM directory has been added.  While the platform CMakeLists.txt files generally only add values to the OT_PLATFORM_DEFINES, they could also look at switches already included to modify behavior.  Likewise, at least one CMakeLists.txt file (simulation) checks to see if OT_CONFIG has been specified -- and in this case, it hasn't, but it WILL be, in the top-level CMakeLists.txt file, about three lines later.

Is this intentional?  If so, why?

If not, then the following change seems to "fix" it for me (it's basically moving the "if (OT_CONFIG)" check and the addition of OT_PLATFORM_DEFINES up about 8 lines) -- I'm happy to open a PR and provide the change if others think it's correct.

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 9605fbf7..1c6e7c6f 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -99,6 +99,13 @@ list(APPEND OT_PUBLIC_INCLUDES ${PROJECT_BINARY_DIR}/etc/cmake)
 list(APPEND OT_PUBLIC_INCLUDES ${PROJECT_SOURCE_DIR}/etc/cmake)
 list(APPEND OT_PUBLIC_INCLUDES ${PROJECT_SOURCE_DIR}/include)
 
+if(OT_CONFIG)
+    list(APPEND OT_PRIVATE_DEFINES "OPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"${OT_CONFIG}\"")
+    message(STATUS "Project core config: \"${OT_CONFIG}\"")
+endif()
+
+list(APPEND OT_PRIVATE_DEFINES ${OT_PLATFORM_DEFINES})
+
 if(OT_PLATFORM STREQUAL "posix")
     list(APPEND OT_PRIVATE_INCLUDES ${PROJECT_SOURCE_DIR}/src/posix/platform)
     add_subdirectory("${PROJECT_SOURCE_DIR}/src/posix/platform")
@@ -107,13 +114,6 @@ elseif(NOT OT_PLATFORM MATCHES "none")
     add_subdirectory("${PROJECT_SOURCE_DIR}/examples/platforms/${OT_PLATFORM}")
 endif()
 
-if(OT_CONFIG)
-    list(APPEND OT_PRIVATE_DEFINES "OPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"${OT_CONFIG}\"")
-    message(STATUS "Project core config: \"${OT_CONFIG}\"")
-endif()
-
-list(APPEND OT_PRIVATE_DEFINES ${OT_PLATFORM_DEFINES})
-
 if(OT_PLATFORM STREQUAL "posix")
     add_subdirectory(src/posix)
 elseif(NOT OT_PLATFORM MATCHES "none")
```

jwhui, 2020-04-21:
@rob-the-dude, thanks for raising this issue! I don't see any reason why it is the way it is. I think it makes sense to move `OT_CONFIG` earlier as you had proposed. Please submit a PR.

abtink, 2020-04-22:
@rob-the-dude , I think this pattern is intentional.

IIRC, the idea is that when `OT_CONFIG` is defined (providing a user given OT config header), it should be used by both platform and OT core. If it is not defined, platform layer has the chance to define its own default config header (by setting `OT_CONFIG` on parent scope).  So the `OT_COFNIG` check and appending to `OT_PRIVATE_DEFINES` happens  after the platform layer from the main CMakefile (to allow platform to set it in this case). 

I think platform also has the chance to add any defines it want (on OT core) by setting `OT_PLATFORM_DEFINES` (on parent scope). I guess that's why `OT_PLATFROM_DEFINES` is added to `OT_PRIVATE_DEFINES` after the platform sub-directory include.

jwhui, 2020-04-22:
@abtink, I wonder if it would be more intuitive the way that @rob-the-dude is proposing. The current model requires the platform to handle `OT_CONFIG` to allow the user to pass down a config header. If a platform doesn't care to set `OT_CONFIG` and simply use default OpenThread config header, then it should not need to deal with `OT_CONFIG` at all. Thoughts?

abtink, 2020-04-22:
If it can work, it sounds good to me.

I also don't mind if the feature that allows the platform to choose a different default config for core is actually removed.

rob-the-dude, 2020-06-14:
A lot of changes have been made to achieve similar results.  I'm going to sit with them a bit and will open a new issue if there's still things I'd like to see added.  I'll also close the PR.

closed 2020-06-14 by rob-the-dude