openthread/openthread #4873: Factory Diagnostics module "Diag"

mamer11, 2020-04-22:
Hello,

I'm trying to activate Factory Diagnostics module "Diag" in CLI, I'm following the example: openthread/examples/platforms/efr32mg21/, I tried  to call OPENTHREAD_CONFIG_DIAG_ENABLE=1 in this way:
"make -f examples/Makefile-efr32mg21 BOARD=BRD4180A OPENTHREAD_CONFIG_DIAG_ENABLE=1" 
I'm not sure if this correct way, I did not get any error, but the "Diag" still not activated, it's not appear in list of available commands after I flash my board, please your advice.

**To Reproduce**
 efr32mg21 

Thanks

bukepo, 2020-04-22:
Could you try the following command instead?
```bash
make -f examples/Makefile-efr32mg21 BOARD=BRD4180A DIAGNOSTIC=1
```

mamer11, 2020-04-22:
Thank you, I tried it but still the same problem. kindly check the attached file, this is part of what I got after I applied the command.
[image]

Thanks

mamer11, 2020-04-22:
Thank you. The problem solved.

closed 2020-04-22 by mamer11