openthread/openthread #4877: SuperframeSpec is limited to recognize 3rd Party BorderRouter Beacon

ghlualven, 2020-04-23:
**Describe the bug**
I have a border router "Kirale KTBRN1" setup for thread network, but my openthread client node cannot recognize the network nor joint. The result is the networkname and xpanid are empty.
```
pi@raspberrypi:~ $ sudo wpanctl scan
   | Joinable | NetworkName        | PAN ID | Ch | XPanID           | HWAddr           | RSSI
---+----------+--------------------+--------+----+------------------+------------------+------
   |       NO | ------ NONE ------ | 0x0745 | 19 | 0000000000000000 | 527DA8551E79EAC9 |  -43
   |       NO | ------ NONE ------ | 0x06B4 | 21 | 0000000000000000 | B7D44072EAD148AF |  -55

| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
> | 0 | OpenThread       | dead00beef00cafe | abcd | 2e6c35c5155ae11a | 11 | -57 | 148 |
| 0 |                  | 0000000000000000 | 0745 | 527da8551e79eac9 | 19 | -61 | 128 |
| 0 |                  | 0000000000000000 | 06b4 | b7d44072ead148af | 21 | -65 | 112 |
Done
```

After captured the data by wireshark sniffer I found the SuperframeSpec from Kirale is not same as Openthread(which is always 0x0ff), how ever the openthread code only allow 0x0ff to pass.
Snapshots attached as well.

I think these bytes should be better handled to allow 3rd party BorderRouter used. I hope this will be fixed ASAP to allow me to use Kirale BR.

```
 enum
    {
        kSuperFrameSpec = 0x0fff, ///< Superframe Specification value.
    };

    /**
     * This method initializes the Beacon message.
     *
     */
    void Init(void)
    {
        mSuperframeSpec     = ot::Encoding::LittleEndian::HostSwap16(kSuperFrameSpec);
        mGtsSpec            = 0;
        mPendingAddressSpec = 0;
    }

    /**
     * This method indicates whether or not the beacon appears to be a valid Thread Beacon message.
     *
     * @retval TRUE  if the beacon appears to be a valid Thread Beacon message.
     * @retval FALSE if the beacon does not appear to be a valid Thread Beacon message.
     *
     */
    bool IsValid(void) const
    {
        return (mSuperframeSpec == ot::Encoding::LittleEndian::HostSwap16(kSuperFrameSpec)) && (mGtsSpec == 0) &&
               (mPendingAddressSpec == 0);
    }
```

ghlualven, 2020-04-23:
[image]
[image]

ghlualven, 2020-04-23:
Screenshot attached.

[image]
[image]

jwhui, 2020-04-23:
@ghlualven, thank you for raising this issue.

For the purposes of joining, Thread prefers using MLE Discovery Request in place of IEEE 802.15.4 Beacon Request. Please try the following wpanctl commane:

```
```

ghlualven, 2020-04-24:
@jwhui thank you for your kindly respond, the command can show the result correctly.

jwhui, 2020-04-24:
> @jwhui thank you for your kindly respond, the command can show the result correctly.

@ghlualven , great thanks!

closed 2020-04-24 by jwhui