openthread/openthread #4895: [posix] netif address handling should be symmetrical

gjc13, 2020-04-27:
**Is your feature request related to a problem? Please describe.** 
In `src/posix/platform/netif.cpp`

```
121 static void UpdateUnicast(otInstance *aInstance, const otIp6Address &aAddress, uint8_t aPrefixLength, bool aIsAdded)
122 {                                                                               
123     struct in6_ifreq ifr6;                                                      
124     otError          error = OT_ERROR_NONE;                                     
125                                                                                 
126     assert(sInstance == aInstance);                                             
127                                                                                 
128     VerifyOrExit(sIpFd > 0, error = OT_ERROR_INVALID_STATE);                    
129                                                                                 
130     memcpy(&ifr6.ifr6_addr, &aAddress, sizeof(ifr6.ifr6_addr));                 
131                                                                                 
132     ifr6.ifr6_ifindex   = static_cast<int>(sTunIndex);                          
133     ifr6.ifr6_prefixlen = aPrefixLength;                                        
134                                                                                 
135     if (aIsAdded)                                                               
136     {                                                                           
137         VerifyOrDie(ioctl(sIpFd, SIOCSIFADDR, &ifr6) == 0, OT_EXIT_ERROR_ERRNO);
138     }                                                                           
139     else                                                                        
140     {                                                                           
141         VerifyOrExit(ioctl(sIpFd, SIOCDIFADDR, &ifr6) == 0, perror("ioctl"); error = OT_ERROR_FAILED);
142     }                                                                           
143                                                                                 
144 exit:                                                                           
145     if (error != OT_ERROR_NONE)                                                 
146     {                                                                           
147         otLogWarnPlat("%s: %s", __func__, otThreadErrorToString(error));        
148     }                                                                           
149     else                                                                        
150     {                                                                           
151         otLogInfoPlat("%s: %s", __func__, otThreadErrorToString(error));        
152     }                                                                           
153 }                                                                               
```

The stack dies when it fails to add an address but only reports an error when it fails to remove it.

**Describe the solution you'd like** 

We may die under both cases.

abtink, 2020-04-27:
IIRC, this is intentional. @bukepo do you remember?
From what I remember, there can be a race situation between add and quick follow-up remove which can cause address remove to seemingly fail.

bukepo, 2020-05-19:
@gjc13 Yes, this is intentional(#4171). But maybe we can enhance this by checking whether the interface is already down, and only removing when it's up.

closed 2020-05-19 by gjc13