openthread/openthread #4897: Periodic parent search unnecessary parent changes

mwilniewiec, 2020-04-27:
From the code analysis it looks to me that periodic parent search can cause unnecessary back and forth parent changes. This would not happen in a stable network but my real life tests are showing that network is hardly ever stable.

If we are below the parent search threshold (-65 by default) we are doing a parent search every 9 minutes (increasing +9min when none better was found).
Better parent is picked basing on multiple comparisons that can change over time
1. link quality
2. number of link 3 connections to other routers
3. number of link 2 connections to other routers
4. number of link 1 connections to other routers
5. RSSI

I would say that there should be an extra comparison called "is a current parent" in the middle to favor current parent over potential one. Probably this comparison must be a bit more complex.

Currently there is a chance that we change to Parent B because its RSSI is -72 instead of -73.
In 9 minutes we will get back to Parent A because it has 9 instead of 8 link 1 neighbors.
In 9 minutes we will go to Parent C because our parent B lost a link 2 neighbor.
In 9, 18 or 27 minutes ....
While searching for any parent - a best pick is obvious, but when searching for a better parent (which has an effect of RLOC change) it should be taken into consideration what is really better for that network.

What are your thoughts?

simonlingoogle, 2020-04-27:
Agree with @mwilniewiec that the Child should favor the current Parent if the benefit of switching Parent does not exceed some threshold.

abtink, 2020-04-27:
Changing the logic for parent selection when device is attached (searching for a better parent) vs. when device is not attached (searching for any parent) sound ok to me.  However, its impact may not be significant, mainly since we don't expect the actual parent switch to happen frequently:

- The 9 minute `OPENTHREAD_CONFIG_PARENT_SEARCH_CHECK_INTERVAL` is how often a child would _check_ whether to perform a parent search or not, i.e., child would check if the running average RSS of its current parent is below a threshold or not. If it is not, the chod does not perform a parent search and checks again in 9 minute.

- If the parent average RSS is below the threshold, we do perform a parent search. After this we don't allow a parent search again within a backoff interval `OPENTHREAD_CONFIG_PARENT_SEARCH_BACKOFF_INTERVAL` (with default value being 10 hours). 

- This backoff wait happens after a parent search, independent of whether the child stayed with its current parent, or found a better parent.

- So a parent switch from "periodic parent search" feature should happen at most every 10 hours.

Note that "parent search" process is rather energy-consuming for an SED (since the child needs to stay in RX mode for few seconds to receive "ML Parent Response" msgs from any potential parents). So we cannot in general perform it often on an SED.

----

There is  bit more logic (which we added later) on how parent search backoff is handled where under certain situation we allow backoff wait time to be canceled (at most once within each backoff interval). 
This was added to handle a practical situation when a router is going through sw update (and may is offline for some time).
- Child is attached to parent A. 
- Child earlier did parent search and is currently in backoff wait mode of 10 hours.
- Device A is going through a sw update (thus it is off for some time)
- This causes the child to detach from A find a less desirable parent B.
- In this situation, we allow the backoff wait time to be canceled, so that after 9 min we can check again and perform a parent search and go back to A.
- Without this, child can stay with less the desirable parent B for up to 10 hours before it does parent search again and find A.
- Note that we allow the canceling of backoff wait to happen only once within a backoff interval.
- It is explained a bit more here in this comment in the `mle.cpp`:
https://github.com/openthread/openthread/blob/a58163c9a312f6504c5a4564ed69a3f4cca97366/src/core/thread/mle.cpp#L4227-L4240

mwilniewiec, 2020-04-28:
@abtink thanks for explanation. Must admit I was more focused on HandleParentResponse and IsBetterParent functions and didn't get the real point of HandleParentSearchTimer.
My previous understanding was that 10 hours backoff is for the situation when our parent is above the threshold. "My parent it good enough, dont bother with parent searching for 10 hours".
9 minutes for a check and 10 hours for a parent search sounds like huge numbers :)
In the 140 nodes network where I make tests I have a feeling that nodes never keep the same RLOC for 10 hours (detailed tests are in progress) so we have bigger problems than this one for sure.

Still ... I believe that there is some room for improvement in this area but changing a parent back and forth in special circumstances every 10 hours sounds like a minor issue.
As long as we keep default DEFINEs its not an issue for me and can be closed. Up to you, thanks.

jwhui, 2020-05-20:
@mwilniewiec , closing issue for now. Feel free to reopen if you have new thoughts.

closed 2020-05-20 by jwhui